j123b567/scpi-parser #20: New entry in scpi_special_number_t suggested

andrey-nakin, 2015-03-15:
CONFigure command accept following values for the 1st arguments: DEF, MIN, MAX, AUTO, <explicit number>. Examples:

CONF:VOLT:DC DEF
CONF:VOLT:DC 12.34
CONF:VOLT:DC AUTO

I'd like to use SCPI_ParamNumber to parse command, but scpi_special_number_t enumeration has no option for AUTO value. Can you add it? Or can I add an exception for this command only to support AUTO option?

Thanks

j123b567, 2015-03-15:
I have implemented better handling of this in experimental branch. You are not forced to use default list of special values but you can define it per parameter.

In the first step, there is nothing against adding AUTO as additional special value.

In the second step, I will try to backport this feature from experimental branch to old version.

andrey-nakin, 2015-03-16:
Thanks for reply. I'll look forward to the next release.

commit f8536dae (2015-03-18): Add new AUTO special value, resolve #20
commit 323ec66d (2015-03-18): Add new AUTO special value, resolve #20
closed 2015-03-18 by j123b567