j123b567/scpi-parser #21: SCPI response after multiple commands in a single packet

andrey-nakin, 2015-03-17:
When I send two or more query commands in a single packet like this:

*OPC?;SYST:ERR?

my Agilent 34401A returns all responses in the single line separated by semicolon:

1;0, "No error"

It is possible to implement this in scpi-parser? Currently command responses are separated by a new line.

j123b567, 2015-03-17:
I will look at this. It will be nice feature.
Nevertheless, I think that standard forbids this at all but I must investigate it little bit.

j123b567, 2015-03-18:
According to the standard, it is possible to call multiple Queries at once.

Multiple responses from one command should be separated by "," which is already done.

Multiple responses should be separated by <RESPONSE MESSAGE UNIT SEPARATOR> - ";" - this will be implemented.

closed 2015-04-21 by j123b567

andrey-nakin, 2015-04-22:
Great fix, thank you!