j123b567/scpi-parser #27: Adding a double-to-string conversion function to scpi_t

andrey-nakin, 2015-04-01:
`SCPI_ResultDouble` function calls `doubleToStr`, which itself calls standard `snprintf` function to convert `double` to string.
On some platforms, e.g. Arduino, standard library does not implement double-to-string conversions, and one has to write his own ones. For example, this code: `printf("v=%lg", 1.23)` actually prints "`v=?`".
I suggest adding a double-to-string conversion function to `scpi_t` type. This function could be optional, i.e. user has not to specify it on platforms where standard library works well.

j123b567, 2015-04-01:
This is the reason why I use intermediate function `doubleToStr`. After some experimenting and searching existing codes I found, that writing custom double to string function is not so easy - it is not easy to do it right.

If it is possible to use `dtostrf`, `gcvt` for arduino, we can create some `#ifdef` for it.

If you have BSD/MIT licensed double to string function like `dtoa`, you can prepare patch for systems without double sprintf and I will be happy incorporate it.

andrey-nakin, 2015-04-18:
I suggest using [`dtostre`](http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__stdlib_1ga6c140bdd3b9bd740a1490137317caa44.html) instead as follows:

    dtostre(<value>, <buffer>, 6, 7)

where 6 - decimal precision, and 7 - bit flags. Displaying value will look like this

    -7.530865E+00
    +3.580245E+00
    +9.900000E+37

Numbers above look like inAgilent 34401A.

j123b567, 2015-04-18:
Can you try latest version if it works for you?

I have found some simple functions to do conversion myself so I will remove sprintf in future version. Nevertheless detection to avr-libc and usage of `dtostrf` will be still useful for AVRs.

j123b567, 2015-04-19:
Pulled #31 but I hope, that it is just temporary fix. All definitions should be in `stdlib.h` file of avr-libc.

j123b567, 2015-04-21:
Remove temporary fix by c804c551d0fee9aacc66b744a443fdbdb9dee12b

Solved now for avr and other systems with floating point sprintf. I will do more work for other systems without floating point sprinf when requested. Closing this for now.

closed 2015-04-21 by j123b567