j123b567/scpi-parser #28: Inappropriate signature of scpi_interface_t.test callback

andrey-nakin, 2015-04-10:
The `*TST?` SCPI command returns `0` if the self-test is successful and `1` if the test fails.

The `scpi_interface_t.test` callback returns value of `scpi_result_t` type with valid values `1` and `-1`. The returned value is directly passed to output buffed by `SCPI_CoreTstQ` function. Seems like a minor data mismatch.

Is it possible to modify `SCPI_CoreTstQ` or change the signature of `scpi_interface_t.test`?

j123b567, 2015-04-10:
You are right. I thing that easiest way is to modify SCPI_CoreTstQ to do someting like

`result = context->interface->test(context) == SCPI_RES_OK ? 0 : 1;`

j123b567, 2015-04-10:
Problem is from #18 and #19 and these changes make everything towards changing prototypes. After a while, I thing that it was not good idea. I must thing about little bit.

j123b567, 2015-04-13:
I hope, that this is final resolution of problems with *TST? command.

closed 2015-04-13 by j123b567

j123b567, 2015-04-13:
Removing this callback in experimental branch completely so user is forced to reimplement whole *TST? callback function with appropriate `SCPI_ResultInt`

andrey-nakin, 2015-04-14:
I'll avoid using `SCPI_CoreTstQ` in my code.
Thank you!