j123b567/scpi-parser #30: Visibility of matchCommand function

andrey-nakin, 2015-04-18:
I found `matchCommand` very useful in my command handlers like `SENSE:FUNC`:

    scpi_result_t sense_function(scpi_t* context) {
      const char* param;
      size_t param_len;

      if (!SCPI_ParamText(context, &param, &param_len, TRUE)) {
        return SCPI_RES_ERR;
      }

      if (matchCommand("VOLTage", param, param_len)) {
        // switch to DC voltage   
      } else if (matchCommand("VOLTage:RATio", param, param_len)) {
        // switch to DC voltage ratio
      } else...`
    }

But visibility of `matchCommand` prevents using it in my code. Could you change its visibility or add a public wrapper function?

Thanks

j123b567, 2015-04-18:
`SCPI_ParamChoice` is not enough for you? It does not support ":" in the name and it can't support it because of IEEE488.2 but it is very similar to your task.

You can define the list by array

```c
const char * sense_function_list[] = {
    "VOLTage",
    "VRATio",
    NULL /* termination of option list */
};

scpi_result_t sense_function(scpi_t* context) {
    int32_t param;
    if (!SCPI_ParamChoice(context, sense_function_list, &param, true)) {
        return SCPI_RES_ERR;
    }

    switch(param) {
        case 0:
            // handle voltage
            break;
        case 1:
            // handle voltage ration
            break;
        default:
            return SCPI_RES_ERR;
    }
    return SCPI_RES_OK;
}
```

And later call

~~~~~~~~~
SENSE:FUNC VOLT
SENSE:FUNC VRAT
~~~~~~~~~

andrey-nakin, 2015-04-18:
Unfortunately `SCPI_ParamChoice` is not an option. Argument can have a colon, e.g. `VOLTage:DC`, and this is standard value. If I simply change it then I lose compatibility with other SCPI devices.

Could you add a preprocessor directive to change visibility?

commit f275a911 (2015-04-18): Export matchCommand by SCPI_Match - resolve #30

matchCommand prototype may change in the future. Making
public API function should resolve future issues.
commit 82e8043c (2015-04-18): Export matchCommand by SCPI_Match - resolve #30

matchCommand prototype may change in the future. Making
public API function should resolve future issues.
closed 2015-04-18 by j123b567