j123b567/scpi-parser #32: Including inc/scpi/config.h in inc/scpi/scpi.h

andrey-nakin, 2015-04-20:
Can you include `inc/scpi/config.h` by `inc/scpi/scpi.h`? I need `SCPI_doubleToStr` macro in my code and I don't want to include anything but `scpi.h`.

j123b567, 2015-04-21:
I will do some changes and renaming. I don't want to make this macro public, but I can make public `doubleToStr` function from `utils.c` with name `SCPI_doubleToStr`. I don't like the situation in the `config.h`, because all macros are badly named.

It will take some time to make these changes.
For you, nothing will change. Just `SCPI_doubleToStr` will not be macro but function. You can use `#include "scpi/config.h"` until I fix this.

andrey-nakin, 2015-04-21:
Thank you for response. I need `doubleToStr` or similar in SCPI handlers like `CONF?`.

j123b567, 2015-04-21:
Sorry, I must change the name of the function. Now it is `SCPI_DoubleToStr` with big "D" to corespond with other names used in the library. I also make public `SCPI_LongToStr`.

commit a2e83e5f (2015-04-21): Make public SCPI_LongToStr, SCPI_DoubleToStr

Make functions SCPI_LongToStr and SCPI_DoubleToStr public. It is
possible to call them from user code. Resolve #32
commit c49b34ae (2015-04-21): Make public SCPI_LongToStr, SCPI_DoubleToStr

Make functions SCPI_LongToStr and SCPI_DoubleToStr public. It is
possible to call them from user code. Resolve #32
closed 2015-04-21 by j123b567