j123b567/scpi-parser #34: Missing declarations

spelunker76, 2015-06-24:
I downloaded the beta version and got it working. I then needed the fixes for #16 (multiple commands). I started doing the patched listed and saw that the parser.c had other changes, so I thought I would try to get the latest. I now have several defines missing. 
c:\tf_workspaces\cxl\utils.cpp(118): error C3861: 'SCPIDEFINE_doubleToStr': identifier not found
1>c:\tf_workspaces\cxl\utils.cpp(158): error C3861: 'SCPIDEFINE_strncasecmp': identifier not found
1>c:\tf_workspaces\cxl\utils.cpp(181): error C3861: 'SCPIDEFINE_strncasecmp': identifier not found
1>c:\tf_workspaces\cxl\utils.cpp(526): error C3861: 'SCPIDEFINE_strnlen': identifier not found

I also have an error:

error C2039: 'paramlist' : is not a member of '_scpi_t'

I think I have the latest files. Any suggestions are appreciated. 

Thanks

j123b567, 2015-06-24:
Please, don't try to patch it. There are much more commits to apply.

Download the latest version instead. Switch to experimental branch on top and use "Download ZIP" button on the right.

j123b567, 2015-06-24:
I have released also new beta version v2.0 beta2 8384701

closed 2015-06-24 by j123b567

spelunker76, 2015-06-24:
Thank You

Bob Schaefer
R&D Project Manager
Component Test Division
Keysight Technologies, Inc.
1400 Fountaingrove Parkway
Santa Rosa, CA 95406-1783
(707) 577-2210

[http://emg.communications.keysight.com/marketing/brand/docs/emailBanners/EmailSig_KST_Transitional-4.png]