j123b567/scpi-parser #35: Error handling for device specific errors

lhoerl, 2015-07-30:
Hi, great work!
I wonder how I can add evice specific errors. Example: A command that sets a float value may be perfectly parsed but the value given is below or above a certain level. In this case I would like to have an extra error code. How can I add these codes? How would be the API?

Currently I work an integrating the libscpi to a Cortex-M4 device with USBTMC. It works and uses about 8k of Flash.

Again, thank you for you work.
Best regards
Lutz

j123b567, 2015-07-31:
There is no API to define custom errors.

## Easy way
- Follow chapters 21.8.9 - 21.8.16 of SCPI-99 specification, and add custom errors to `libscpi/inc/scpi/error.h`. Prefere already existing errors in the spec.
- If you can make your changes open, create pull request and I will be happy to incorporate them to mainline.

## Extended way
 - implement api to define custom errors and I will be also hapy to incorporeate this to mainline

lhoerl, 2015-08-03:
Hi again, as I am a newbie on GIT I can only hope that I did it the right way:
I checked out a 'fork' of the libscpi project, worked on it and 'pushed' it back to the server.
Now I can access it via web on:
https://github.com/lhoerl/scpi-parser

I think until now I did not understand correctly the idea behind 'fork' versus 'branch' but anyhow as far as I understand it now I should sens a 'pull request'?
best regards
Lutz

koeart, 2015-08-03:
Hey,

> Hi again, as I am a newbie on GIT I can only hope that I did it the
> right way: I checked out a 'fork' of the libscpi project, worked on
> it and 'pushed' it back to the server. Now I can access it via web on:
> https://github.com/lhoerl/scpi-parser
Yes.

> 
> I think until now I did not understand correctly the idea behind
> 'fork' versus 'branch' but anyhow as far as I understand it now I
> should sens a 'pull request'? best regards Lutz

Correct. The difference between a fork and a branch:
A branch is done "in the same project", so basically if you would have
permission to push to j123b567's scpi-parser git, you would have done a
branch, tested it and then merged it back in the master branch.

Because you don't have the necessary permissions, you fork the project,
work on it (e. g. do a branch, test your changes, merge the branch back
to master if successful) and you can send a pull request with the
changes. What happens: You request j123b567 to "pull" a commit from
your fork and merge it into his project.

A probably better description can be found here:
<https://help.github.com/articles/using-pull-requests/>

BTW, thanks for the work! I have to use the scpi-parser for an Embedded
Project (ARM AT91SAM7) as well, so thanks in advance for sharing your
work! I hopefully will have something to contribute to this project as
well, soon.

Best regards,
    koeart

j123b567, 2015-08-03:
@lhoerl Thank you for the pull request. Further discussion about your changes will be there.

@koeart Happy to hear this! :)

j123b567, 2015-08-04:
@lhoerl I have change your proposal little bit I think you can migrate your project easily. I'have introduced define `SCPI_USER_CONFIG` which should be defined in project. After this change, you can create your own file `scpi_user_config.h` In this file, you should override `USE_FULL_ERROR_LIST`,  `USE_USER_ERROR_LIST`, `SCPI_LINE_ENDING` and define `LIST_OF_USER_ERRORS`. Please **don't include any `scpi` header in this file**, because it will cause undefined behaviour.

Content of `scpi_user_config.h` should be for example

    #define USE_FULL_ERROR_LIST 1
    #define USE_USER_ERROR_LIST 1
    #define SCPI_LINE_ENDING "\n"
    #define LIST_OF_USER_ERRORS \
        X(SCPI_USER_ERROR_1, 1, "User error 1") \
        X(SCPI_USER_ERROR_2, 2, "User error 2")

Can you please test it and tell me if everything is ok for you? After that I can release new version.

j123b567, 2015-08-05:
It seems, that it works, so closing this issue.

closed 2015-08-05 by j123b567