j123b567/scpi-parser #38: `Error queue not empty` bit not set

j123b567, 2015-08-06:
@lhoerl Please can you write here more info about this.

lhoerl, 2015-08-06:
Look into SCPI 1999, Chapter 9, Status Reporting, page 9-2, the diagramm printed there.
You can see a line from the error queue to bit 2 of STB. This bit should indicate "error queue not empty". Also there is bit 4 MAV that should be set when there is a Message AVailable.
[image]

The MAV bit is only 'visible' as set when you split a query, i.e. send e.g. *IDN? to a device but don't read immediately. In this case when you read out the STB via a status byte request (IEEE488.1 or USBTMC) and not via standard read/write interface the MAV should report "OK, you can read now, there is a message available".

In USBTMC a device can be setup so that it can generate an interrupt that is forwarded to the host via an extra pipe. The standard commands and responses are transferred using BULK OUT and BULK IN pipes. That enables the device to send a request to the host (PC) which is normally not possible when the PC is the master (of communication). The device would hacve to wait until it is asked by the host to report an 'event' or 'status'. With USBTMC class an interrupt transfer is used and with the appropriate driver on the host side the device triggers a callback function on the host. Looking deeper we realize that this mechanism needs of course the help of some low level drivers on the PC side that poll the USBTMC interrupt pipe regularly. I think the mechanism is similar to HID devices (keyboard) when you hit a key.

regards Lutz

j123b567, 2015-08-06:
Thank you. I think, I can correct bit 2 "Error queue not empty"

After some time of thinking, setting MAV is up to you. In raw TCP or serial communication, there is no meaning  for this, because message is directly send without waiting.

On USBTMC, GPIB and possibly on VXI-11, HiSLIP, there is a reason for this and you can easily implement it. After libscpi calls WRITE callback, set MAV to 1. After your output buffer is emptied by the user, set MAV to 0. I can set MAV to 1 in the library, but I don't know when to set it to 0, so it is better not to touch MAV in the library at all.

lhoerl, 2015-08-06:
Great!
Please confitm that the use of
void SCPI_RegSetBits(scpi_t * context, scpi_reg_name_t name, scpi_reg_val_t bits);
void SCPI_RegClearBits(scpi_t * context, scpi_reg_name_t name, scpi_reg_val_t bits);
the right way of setting/clearing the MAV bit in STB?
best regards
Lutz

j123b567, 2015-08-06:
Yes

    
    // set MAV
    SCPI_RegSetBits(&scpi_context, SCPI_REG_STB, STB_MAV);
   
    // clear MAV
    SCPI_RegClearBits(&scpi_context, SCPI_REG_STB, STB_MAV);

commit 6e5e3e0e (2015-08-06): Resolve #38 Error queue not empty bit
closed 2015-08-06 by j123b567