j123b567/scpi-parser #39: get unsigned integers with SCPI_ParamNumber

bitwigglers, 2015-08-10:
Is it  possible with the current API to retrieve an unsigned integer value from a numeric parameter like "#hffffffff"?

bitwigglers, 2015-08-10:
SCPI_ParamDouble can be used, since a double can represent all 32 bit integers.

j123b567, 2015-08-10:
This will not solve your problem. I should extend the API to support this.

j123b567, 2015-08-10:
If the number is HEX, BIN or OCT, it is always converted by `SCPI_ParamToInt` - with sign and then converted to double in `SCPI_ParamDouble` - still preserve the sign and still wrong number.

bitwigglers, 2015-08-10:
You are right. I just saw that SCPI_ParamToDouble() calls SCPI_ParamToInt() for non-decimal numbers.

j123b567, 2015-08-10:
You can now use `SCPI_ParamUnsignedInt`, `SCPI_ParamDouble` or `SCPI_ParamNumber`. All should work as expected.

closed 2015-08-10 by j123b567

bitwigglers, 2015-08-10:
Wow, that was a quick fix. Will test it asap. Thank you very much!

commit deae06da (2015-08-10): Resolve #39: Correct handling of unsigned integers