j123b567/scpi-parser #42: SCPI_ParamText() does not correct the paramlist.length if text is "-capsulated

mk-at-sch, 2015-08-27:
Not correcting the length will result in "Parameter not allowed error" (more parameters than allowed) - but I'm not on the current code base...
The text parsing is correct, but the tests do not check for the length (== 0) after processing, correct?

I did a quick fix for my needs to detect the "" and correct the length:
```c
// ...
if (locateText(context->paramlist.parameters, 
            context->paramlist.length, value, &length)) {
    *// Work-around: Correct paramlist.length
    if(*context->paramlist.parameters == '"')
        paramSkipBytes(context, 2);

    paramSkipBytes(context, length);
    if (len) {
        *len = length;
    }
    return TRUE;
}
```
Thanks.

j123b567, 2015-08-27:
Can you please show me some example, how to reproduce this error? Example scpi command with parameters and how you handle it.

mk-at-sch, 2015-08-27:
Perhaps it's due to the old code base. This function was unchanged since then, but perhaps the handling of remaining bytes after processing the expected parameters was modified.
I could reproduce this by using the SCPI_ParamText function with a text in "".
I just found out that locateText is tested isolated but there is no test for SCPI_ParamText (TODO flag), am I correct?
If you add a sample command expecting one text parameter and test it with a text w/o and w/ "" then an error should be produced for the test in "".

My handling in the called function is just 
```c
// Read string
if(SCPI_ParamText(context, &cStrPtr, &len, FALSE)) {
    if(len > XY_STRINGSIZE - 1) {
        SCPI_ErrorPush(context, SCPI_ERROR_DATA_OUTOFRANGE);
        return SCPI_RES_ERR;
    }
}
```

j123b567, 2015-08-27:
OK, there is no longer function `SCPI_ParamText`, because it does not work as expected. If you don't care about `"` or `'` inside of the string, you can now use `SCPI_ParamCharacters`. If you care, you must use `SCPI_ParamCopyText`. Both functions work properly and they are also well tested in `test_parser.c`.

If you have same problems with newest version, just open new issue, but I must close this as invalid, because it is no longer true.

closed 2015-08-27 by j123b567

mk-at-sch, 2015-08-28:
You are correct - sorry for that. I mixed up different sources. My fault.
Thanks for the hint.