j123b567/scpi-parser #45: Usage of scpi_special_numbers_def[]

koeart, 2015-09-02:
Hello,

I'm a bit puzzled how you can setup special numbers for your instrument?
As I found out, there is ```scpi_special_numbers_def```, in which all necessary ```MINimum, MAXimum, ...``` defined.
If I use ```ParamNumber( ... )``` to parse the value, how can I substitute now my DEFault, MINimum and MAXimum?
I would imagine something like this:
```
struct instrument_voltage {
    scpi_number_t my_default;
    scpi_number_t my_min;
    ...
};
```
But I think I just oversaw something in the lib or didn't understand something correctly. An Input string which could be used from the commandline is, for example:
```conf:volt:dc Default, 1.4 mV```
 Thx in advance!

j123b567, 2015-09-02:
Hello,
it is little bit tricky. You can take some inspiration in `SCPI_NumberToStr`.

`scpi_number_t` type has this fields
  - `special` means, that the number is not a number, but MIN, MAX, DEFAULT, 
  - `tag` is the index to `scpi_special_numbers_def` or other array of definitions and is valid only if `special==TRUE`. For `scpi_special_numbers_def`, there are set of defines `SCPI_NUM_MIN`, `SCPI_NUM_MAX`, ... but you can use your own list with your own values.
  - `value` is value in base units and is valid only if `special==FALSE`
  - `unit` is unit of `value` if available
  - `base` is the base that the number had prior any processing. You can see here, if the user typed the number in base DEC, HEX, BIN ...  

Is this what you want?

koeart, 2015-09-02:
Well, I looked into ```scpi_number_t```. Thanks for explaining it in more details. I think it still didn't answer my question.

Assume my voltage ```DEFault``` = 23.42 V.
Now I could run (from the examples/test-interactive):
```conf:volt:dc 23.42 V```
I can now read this with ```SCPI_ParamNumber(context, scpi_special_numbers_def, &param1, False
```.
If I want to access this variable I use
```
double myvoltage = param1.value;
```
What if someone enters
```
conf:volt:dc DEF
```
During parsing I understand that your parser accepts DEF as a value, but how does it know the value? How can I define this? It would be nice to have something like ```special_numbers_voltage```, ```special_numbers_current```, ... This I would pass to ```SCPI_ParamNumber( ... )```.
Hope it's clear what I mean?

j123b567, 2015-09-02:
If someone enters `conf:volt:dc DEF` you must do it yourself.

```c
res = SCPI_ParamNumber(context, scpi_special_numbers_def, &param1, FALSE);
if (!res || (param1.special && param1.tag==SCPI_NUM_DEF)) {
    param1.special = FALSE;
    param1.value = 23.42;
    param1.unit = SCPI_UNIT_VOLT;
}
// continue with regular processing of param1
double myvoltage = param1.value;
```

You need something like this?

```c
const scpi_choice_def_t special_numbers_voltage[] = {
    {"DEFault", SCPI_NUM_DEF, 23.42, SCPI_UNIT_VOLT},
    SCPI_CHOICE_LIST_END,
};
```
So `SCPI_ParamNumber` can handle it without additional code? If so, what to do with `UP`, `DOWN` and `AUTO` and possibly others computed values. Other values may be context specific and may depend on other values. For one range, `DEF` is something and for other range, it is completely different value. Because of this complexity, I don't like to put this functionality to `SCPI_ParamNumber`.

Is this what you mean, or am I still wrong?

koeart, 2015-09-03:
No, it answers my question perfectly and your argument makes sense. Thanks!

closed 2015-09-03 by j123b567