j123b567/scpi-parser #46: Question: Error vs not present for optional parameters

mvladic, 2015-09-14:
When calling, for example, `SCPI_ParamInt` with argument `mandatory` set to `FALSE`, how to differentiate real errors from the case when parameter is not present (in both cases the return value from this function call will be `FALSE`)?

koeart, 2015-09-14:
Have a look at https://github.com/j123b567/scpi-parser/blob/master/libscpi/src/parser.c#L653.
How ```SCPI_ParamInt``` works is, you copy your parameter to a variable (see https://github.com/j123b567/scpi-parser/blob/master/examples/common/scpi-def.c#L100 for an example), it just returns TRUE (Success) or FALSE (otherwise) if the parameter wasn't an Integer value.

The function SCPI_ParamInt will return
TRUE if successful and
FALSE in case the Datatype doesn't match or there was a parameter with a suffix (one should handle this with ```SCPI_ParamNumber```).

You have to check the integer value afterwards yourself (for example if it is in the correct range or whatever).

Thats why there is an
```
if (!SCPI_ParamInt( context, &param, FALSE)) {
     param = 23; //some default value;
}
```

Or did I miss something there?

j123b567, 2015-09-14:
@koeart I think that @mvladic has different problem and he is right that there is no souch direct functin for it. There are some solutions more or less ugly.

- split parameter handling into two functions `SCPI_Parameter()` and `SCPI_ParamToInt()`. Result of `SCPI_Parameter()` tells you, if parameter was here or not. You can later check the type of the parameter. If the function returns `FALSE` and parameter type is `SCPI_TOKEN_PROGRAM_MNEMONIC`, this means missing parameter. If type is `SCPI_TOKEN_UNKNOWN`, error occured. If `SCPI_ParamIsNumber()` then it is safe to convert it to integer.
- Before and after calling `SCPI_ParamInt()`, you can readout `SCPI_ErrorCount()`. If the number changes, it meas, there was an error.
- Do some magic in `error()` callback and set some global variable to detect this.

I don't want to break current API so there are some solution for the future:

- introduce something like `errorno`. There will be last error of the `SCPI_ParamXYZ()` function or 0. 
- introduce API function`SCPI_ParamIsInvalid()` so you can check parameter from `SCPI_Parameter()` without relaying on some parser internals.

What do you think?

mvladic, 2015-09-14:
I like `errorno` solution more because I don't need to split parameter handling into two functions (`SCPI_Parameter()` and `SCPI_ParamToInt()`). In that case I will do the following:

    if (!SCPI_ParamToInt(context, ...)) {
        if (errorno) {
            return SCPI_RES_ERR;   
        }
        // not a real error, i.e. parameter is missing, continue
    }

You could also add SCPI_ParamIsInvalid(), it looks like an useful function.

j123b567, 2015-09-14:
Sounds good. I will extend `context` with `errno` and implement `SCPI_ParamIsInvalid()`.

This is almost mechanical but large change so it will take some time to implement this. I would like to add also unit tests for this to be sure it is correct. So please be patient.

mvladic, 2015-09-14:
Thanks! In the meantime I will use one of the workarounds that you proposed.

j123b567, 2015-09-14:
It was easier than I thought. The functionality was already there. 

You can use 

```c
    if (!SCPI_ParamToInt(context, ...)) {
        if (SCPI_ParamErrorOccurred(context)) {
            return SCPI_RES_ERR;   
        }
        // not a real error, i.e. optional parameter is missing, continue
    }
```

commit c1e514e0 (2015-09-14): Resolve #46: Detect missing parameter
closed 2015-09-14 by j123b567