j123b567/scpi-parser #49: Trouble w/ SCPI_ResultDouble

koeart, 2015-09-24:
Hello,

I'm developing an embedded ARM Project (ARM7TDMI, Atmel AT91SAM7X128). While integrating SCPI told me about issues with some newlib stuff (which I could solve), I'm now in trouble when trying to display `double` numbers. Doing multiplactions with them is not a problem, though.

My question: What standard libraries do you rely on? Do you maybe have a suggestion where I can look, what to fix? Unfortunately I cannot connect via JTAG to the controller (and don't have access to UART yet).

I will do a workaround for the time being, but I would be happy if I could fix this later. I'm also looking into other ressources, but maybe there is a hint/fix available for this I think could be a common issue on embedded systems without native floating point support.

Thx in advance!

P.S.: `SCPI_ResultDouble` and `SCPI_ResultText` (in conjunction with `SCPI_NumberToString`) crash.

j123b567, 2015-09-24:
Conversion from string to double is easy and it uses `strtod` but there is no good way to convert double to string.

On AVR, I'm using `dtostre` on other platforms, `snprintf(...., "%lg", ...)` is used. `snprintf` has big disadvantage that it uses `malloc`/`free` and it needs lot of heap memory - start at least with 4kB. If you don't have working `malloc`, or your heap is smaller then 4kB, it will fail.

In `config.h` there is define `SCPIDEFINE_doubleToStr(v, s, l)` where `v` is double value, `s` is result string and `l` is a length of provided string buffer. Return value should be length of result string. If you have better implementation of doubleToStr, you can just redefine this macro.

We can start with http://stackoverflow.com/questions/7228438/convert-double-float-to-string there is a link to musl-libc library, which is licensed under MIT licese (so compatible with us). And it is possible to extract double to string conversion from there.

j123b567, 2015-10-05:
I would like to implement some minimal and not so precise implementation of this, so we can live without `sscanf` on limited platforms.

j123b567, 2015-11-10:
I added new function in c51c359b7601f4f071444a4579d1678b5aac7d06. It is based on old FreeBSD's implementation of `fcvt`. If you define `USE_CUSTOM_DTOSTR` as 1 in `config.h`, it will use this function in double and float to string conversion functions. It is still not perfect, because it needs 310 bytes on stack. I would like to minimize this, but I don't have much time to study the conversion algorithm.

It is still better for minimal systems, because `sprintf` needs about 4k on heap and needs `malloc`.

closed 2016-02-20 by j123b567