j123b567/scpi-parser #50: missing shared library

jeras, 2015-09-29:
building a shared library was removed in the next commit:
https://github.com/j123b567/scpi-parser/commit/6e73d8531ba409e513c2d491d473a7b593614733

We are attempting a migration from v1.2 to v2.1 and we are missing the shared library. Could you explain your reasoning behind this removal?

Support for shared libraries would be needed for example if libscpi was packaged for Debian or another distribution.

j123b567, 2015-09-29:
This library primarily targets bare metal systems using its own build system. The reason why are there any Makefiles are unit tests.

Building of shared library was broken and unmaintained. It was just a proof of concept but I never use it and I didn't test it if it at least works.

There is also missing `install` phase.

Supporting full blown systems like Linux was not a goal, but now it is good time to do it.

If you have some patches to Makefile or autotools/CMake/whatever build tool files, I can merge them, just create a pull request.

RedPitaya, 2015-09-29:
Thanks for the explanation.

I am not very good at writing Makefiles, so it will take me some time to write a reasonably good version and test it, but I already have an ugly version in a fork. We will start testing it soon, but it will take us some time.
https://github.com/RedPitaya/scpi-parser

For now I can tell you the shared library in version v1.2 was was compiled properly with the provided Makefile, at least using a cross compiler for ARM.

j123b567, 2015-09-29:
Ok, I can investigate it little bit.

btw, nice hardware :)

RedPitaya, 2015-09-30:
The Makefile fix from RedPitaya fork (actually reverted commit which removed shared lib support) works. But is not well structured, and -fPIC is used two times, due to some haste copy-paste-ing.

Actually I expected more trouble with porting our patches from 1.2 to 2.1, and was surprised when it worked.

j123b567, 2015-10-01:
I tried to fix this in 9196799, can you please test/comment it?

RedPitaya, 2015-10-02:
I tested it.

Compile seems to work fine, both a static and a dynamic library are created. Sources are compiled twice, once for the static lib and once for the dynamic, I do not know if this is how it should be.

But install performs another compile, which in my case fails, since I do not define the cross compiler for this step.

j123b567, 2015-10-02:
It is compiled twice because of dynamic is compiled with -fPIC but static not.

It is strange, that install performs another compile step. I will investigate this.

j123b567, 2015-10-02:
Ok, next try 3074d4b. Issue with install should be ok now.

RedPitaya, 2015-10-02:
Thanks, I already tried it and it works fine.

I also checked if it is possible to avoid recompiling object files for shared and static libraries. Apparently it is possible but not recommended.

I have no further comments now. This issue can be closed.

closed 2015-10-02 by j123b567