j123b567/scpi-parser #55: How to  use SCPI_ExprChannelListEntry()?

koeart, 2015-10-06:
Hi,

I want to use the channel list capability you introduced in commit 7dca581.

But how to use/parse it properly?

For example, how to parse
`:READ? (@1:2!4)` (Reading Channels 1, 2, 4).

Thanks :)

j123b567, 2015-10-06:
I think, that your example is wrong, because `1` has 1 dimension and `2!4` has 2 dimensions and there is not possible to create range `:` between different number of dimensions.

I will update documentation for this soon.

koeart, 2015-10-06:
Hi,
I completely had the syntax wrong. Should've been `READ? (@1,2,4)`. A better example would be `MEAS:VOLT (@1,3,4:6)`, taken from SCPI99, Vol 1 Ch. 8.3.2 "Expressions - Channel Lists".

j123b567, 2015-10-07:
OK, I don't have much time to write better documentation, so here are just small hints.

In handler of your command, read channel list as parameter using `SCPI_Parameter`.

~~~c
scpi_parameter_t channel_list_param;

// get channel list
if (SCPI_Parameter(context, &channel_list_param, TRUE)) {
    scpi_expr_result_t res;
    scpi_bool_t is_range,
    int32_t values_from[4];
    int32_t values_to[4];
    size_t dimensions;

    // this is valid and it gets real number of dimensions for forst item (index 0)
    res = SCPI_ExprChannelListEntry(context, &channel_list_param, 0, &is_range, NULL, NULL, 0, &dimensions);

    // this will fill values
    // if is_range, then both values_from and values_to are filled
    // dimensions represents number of items in this array. This number is always
    //    number of dimensions and can be larger then size of provided array, but provided array
    //    is filled only up to its size.
    res = SCPI_ExprChannelListEntry(context, &channel_list_param, 0, &is_range, values_from, values_to, 4, &dimensions);
}
~~~~

You must iterate over all entries in the expression by changing index and chacking the result. If result is `SCPI_EXPR_OK` then all values are valid. If the result is `SCPI_EXPR_ERROR`, there was an error and you should stop handling this command.
If the result is `SCPI_EXPR_NO_MORE` then there are no more Entries in the list.
    
For channel list `(@1,3,4:6)`, this will be result
 - for index 0, `is_range = false`, `values_from[0] = 1`, `dimensions = 1`
 - for index 1, `is_range = false`, `values_from[0] = 3`, `dimensions = 1`
 - for index 2, `is_range = true`, `values_from[0] = 4`, `values_to[0] = 6`, `dimensions = 1`
 - for index 3, result is `SCPI_EXPR_NO_MORE`

koeart, 2015-10-11:
I coded a rather long example which handles different cases for channel numbers.
It works for (up to) 2 dimensions so far.
SCPI Vol says to parse ranges in order.
One enters ranges like this: `(@rs!cs:re!ce)`, concatenating of ranges does work, too.
rs means "row start", re "row end", cs "column start", ce "column end".
Examples:
`(@1!1:3!2)` would be 1!1, 1!2, 2!1, 2!2, 3!1, 3!2.
`(@3!1:1!3)` would be 3!1, 3!2, 3!3, 2!1, 2!2, 2!3, ... 1!3.

In this example the whole channel list is written to `array[]`  of dimensions `maxrow` x `maxcol`. Last values of array are {0, 0}. I introduced a datatype at the beginning to hold a value for row and column.

Hope this helps someone. Suggestions welcome!

```
struct _scpi_channel_value_t  {
   int32_t row;
   int32_t col;
};
typedef struct _scpi_channel_value_t scpi_channel_value_t;

/**
 * @brief 
 * parses lists
 * channel numbers > 0.
 * no checks yet.
 * valid: (@1), (@3!1:1:3,4), ...
 * 
 * @param channel_list channel list, compare to SCPI99 Vol 1 Ch. 8.3.2
 */
static scpi_result_t TEST_Chanlst ( scpi_t *context )
{
    scpi_parameter_t channel_list_param;
    #define maxrow 2    //maximum number of rows
    #define maxcol 6    //maximum number of columns
    #define maxdim 2    //maximum number of dimensions
    scpi_channel_value_t array[maxrow * maxcol]; //array which holds values in order (2D)
    size_t chanlst_idx; //index for channel list
    size_t arr_idx;     //index for array
    size_t n, m = 1;        //counters for row (n) and columns (m)
    
    // get channel list
    if ( SCPI_Parameter ( context, &channel_list_param, TRUE ) ) {
        scpi_expr_result_t res;
        scpi_bool_t is_range;
        int32_t values_from[maxdim];
        int32_t values_to[maxdim];
        size_t dimensions;
        
        bool for_stop_row = false;      //true if iteration for rows has to stop
        bool for_stop_col =false;       //true if iteration for columns has to stop
        int32_t dir_row = 1;            //direction of counter for rows, +/-1
        int32_t dir_col = 1;            //direction of counter for columns, +/-1
        
        // the next statement is valid usage and it gets only real number of dimensions for the first item (index 0)
        if(!SCPI_ExprChannelListEntry ( context, &channel_list_param, 0, &is_range, NULL, NULL, 0, &dimensions )) {
            chanlst_idx = 0;    //call first index
            arr_idx = 0;        //set arr_idx to 0
            do {        //if valid, iterate over channel_list_param index while res == valid (do-while cause we have to do it once)
                res = SCPI_ExprChannelListEntry ( context, &channel_list_param, chanlst_idx, &is_range, values_from, values_to, 4, &dimensions );
                if (is_range == false) {        //still can have multiple dimensions
                    if (dimensions == 1) {
                        //here we have our values
                        //row == values_from[0]
                        //col == 0 (fixed number)
                        //call a function or something
                        array[arr_idx].row = values_from[0];
                        array[arr_idx].col = 0;
                    } else if (dimensions == 2) {
                        //here we have our values
                        //row == values_fom[0]
[... 51 lines cut ...]
                            arr_idx++;
                        }
                        if (n == values_to[0]){
                            //endpoint reached, stop row for-loop
                            for_stop_row = true;
                        }
                    }
                    
                    
                } else {
                    return SCPI_RES_ERR;
                    break;
                }
                //increase index
                chanlst_idx++;
            } while ( SCPI_EXPR_OK == SCPI_ExprChannelListEntry ( context, &channel_list_param, chanlst_idx, &is_range, values_from, values_to, 4, &dimensions ) ) ;
            //while checks, whether incremented index is valid
        }
        //do something at the end if needed
        array[arr_idx].row = 0;
        array[arr_idx].col = 0;
    }
    return SCPI_RES_OK;
}
```

j123b567, 2015-10-12:
Nice example. I will integrate this to documentation and to examples.

Maybe, there can be some helper function for 1 dimension and 2 timensions to do this for you.

commit ee12be82 (2015-10-12): Resolve #55: SCPI_ExprChannelListEntry example
closed 2015-10-12 by j123b567