j123b567/scpi-parser #59: SCPI_Input() and input buffer overrun

mvladic, 2015-10-15:
I think `SCPI_Input()` doesn't handle input buffer overruns properly. It checks for overrun and returns -1 if it happens. But, what can I do next? If I continue to feed `SCPI_Input()` with the data it will continue to return -1 and basically block the SCPI parser. I can't reset the buffer, except if I call `SCPI_Input()` with 0, but in that case it will call `SCPI_Parse()` with partially received data.

If buffer overrun is detected in `SCPI_Input()` I think it should generate `-363, "Input buffer overrun"` error, ignore everything until \r\n and reset the input buffer after that so it can start fresh again.

j123b567, 2015-10-15:
You are right, that it should generate -363. Problem is, that only way, how to detect end of command is full blown parse. Arbitrary data can contain false positive `\r\n`.
Because of that, I can reset input buffer in that case and forget everything including current input. No search of end of command.

mvladic, 2015-10-15:
I agree. Thanks!

j123b567, 2015-10-15:
I must change return value of `SCPI_Parse` and `SCPI_Input` to `scpi_bool_t`. It make more sense.

commit e816026f (2015-10-15): Resolve #59: input buffer overrun handling
closed 2015-10-15 by j123b567