j123b567/scpi-parser #63: Status Register update from threads

lhoerl, 2015-11-02:
Hi Jan, maybe you can guide me a little through this special aspect.

In my code I run a RTOS (CortexM with Keil RTX). Some of the threads running in the system produce status information (e.g. temperature too high) that I want to report in the questionable status. Just using the SCPI_RegSet() family of functions may lead to multiple acces to the same variable from different threads, here the USB thread and the temperature thread.

Normally I would protect the access with a mutex and I think I will introduce one here.
Can you give a hint where the mutex is best placed?
Is this a common issue that should be noticed in the library?
Can we think of providing a 'shileding function' that does mutexing like the function pointers in scpi_interface?

Any hint is welcome, best regards, Lutz

j123b567, 2015-11-02:
It is not special request but it is currently not done. Library is currently not thread safe. You should handle everything in one thread. 

Best way is to use message queue. In USB thread, introduce message queue. If you need to handle interrupt of USB, just send some message to this queue. If you need to set register, just send another message to this queue. Wait on this queue in USB thread and process messages.

I can introduce some locking in the future but it is not easy and I don't have much time now for that.

j123b567, 2015-12-05:
Concentrate all related issues to one, so closing this.

closed 2015-12-05 by j123b567