j123b567/scpi-parser #65: Non-ASCII character in string param

mvladic, 2015-11-12:
SCPI parser is blocking if there is non-ASCII character inside string param.

mvladic, 2015-11-13:
I tried your fix and now it is different but still doesn't work. First time I send command with non-ascii character there is a following assertion:

[image]

After that, no matter what command I send I'm getting the same assertion and nothing is executed.

j123b567, 2015-11-13:
I can't reproduce this. After sending non ASCII character - e.g. éščř, it just tell me `ERROR: -101, "Invalid character"` and after that, I can continue and type new command.

Can you do some backtrace at the point of assertion, so we can see what code it does?

Can you tell me, what is `sizeof(char)` in your compiler?

mvladic, 2015-11-13:
Yes, you are right! This problem only occurs while debugging on Windows using my simulator code. In all other cases it works as expected. You can close this issue.

j123b567, 2015-11-13:
OK, but it is still a problem of Microsoft VIsual C. This should be supported compiler so I would like to fix this. I don't have windows, so it is currently not easy for me to test this, so any help or hint is welcome.

mvladic, 2015-11-13:
Here is the backtrace at the point of assertion:

    ucrtbased.dll!0f0e1e06()	Unknown
    [Frames below may be incorrect and/or missing, no symbols loaded for ucrtbased.dll]	
    [External Code]	
    >  eez_psu.exe!skipProgramMnemonic(_lex_state_t * state) Line 308	C
    eez_psu.exe!skipCompoundProgramHeader(_lex_state_t * state) Line 379	C
    eez_psu.exe!scpiLex_ProgramHeader(_lex_state_t * state, _scpi_token_t * token) Line 420	C
    eez_psu.exe!scpiParser_detectProgramMessageUnit(_scpi_parser_state_t * state, char * buffer, int len) Line 1396	C
    eez_psu.exe!SCPI_Input(_scpi_t * context, const char * data, int len) Line 354	C
    eez_psu.exe!scpi_psu_input(_scpi_t & scpi_context, char ch) Line 103	C++
    eez_psu.exe!main_loop() Line 38	C++
    eez_psu.exe!main() Line 7	C++
    [External Code]	

If there is no assertion (for example when I start the code without debugger) everything works perfectly fine! Honestly, I don't think that `assert(c>=-1 && c <=255)` makes sense in `isalpha`.

The reason why in debugger (even if I select to ignore assertion) I'm experiencing the following problem:

> After that, no matter what command I send I'm getting the same assertion and nothing is executed.

is very strange (as usual in Windows world). My simulator code feeds scpi_parser with characters by sending messages from one thread of execution (console input thread) to the another (scpi parser thread). After this assert dialog box shows up, this mechanism stops to work and I don't feed scpi parser with characters any more and that is the reason why nothing is executed event though I'm sending commands from console input.

j123b567, 2015-11-13:
I think I now know, what is wrong. I can fix this soon. Thank you.

commit d1aa0fff (2015-11-12): Resolve #65: Infinite loop with non ASCII character
commit b5e8e4dc (2015-11-13): Resolve #65: Cast char to unsigned in isctype functions
closed 2015-11-13 by j123b567