j123b567/scpi-parser #66: Error queue

mvladic, 2015-11-13:
Some questions and issues with error queue:

- Why is the error queue shared between scpi context's? I was expecting that each context has its own unique error queue.
- There should be a way to configure the size of error queue without changing `FIFO_SIZE` in `fifo_private.h`, preferably with `scpi_user_config.h`.
- If the queue overflows, scpi-parser doesn't behaves according to the SCPI specification which says (paragraph 21.8.1 The Error/Event Queue):

> As errors and events are detected, they are placed in a queue. This queue is first in, first out.
If the queue overflows, the last error/event in the queue is replaced with error
-350,"Queue overflow"
Any time the queue overflows, the least recent errors/events remain in the queue, and the
most recent error/event is discarded.

j123b567, 2015-11-13:
I have separated the specification issue to another report to help me sove this part by part.

You are right at all points. It was the first naive implementation from early versions and nobody complain about it until now :)

mvladic, 2015-12-15:
Something is not quite right here. Take a look at this sequence of commands (queue size is 3):

(Note that every time error is emitted (interface->error callback is called) I print "**ERROR" on stdout.)

```
> SYSTem:ERRor?
0,"No error"
> SYSTem:ERRor:COUNt?
0
> volt 1amp
**ERROR: -131, "Invalid suffix"
> none
**ERROR: -113, "Undefined header"
> SYSTem:ERRor:COUNt?
2
```

Until now everything is OK. But, this is unexpected:

```
> @
**ERROR: -101, "Invalid character"
```

I was expecting: **ERROR: -350,"Queue overflow" because queue size is 3.

Further:

```
> SYSTem:ERRor:COUNt?
2
```

I was expecting 3.

Here is what was on the error queue at this point:

```
> SYSTem:ERRor?
-131,"Invalid suffix"
> SYSTem:ERRor?
-350,"Queue overflow"
> SYSTem:ERRor?
0,"No error"
```

j123b567, 2015-12-15:
I have bug in queue implementation. Now it needs N elements in buffer to hold (N-1) items. So size of error queue initialized with 3 should be 2.

I will try to reproduce your beahivour and correct potencial error.

j123b567, 2016-02-13:
Queue length is now intuitive again.

Error callback is now colled twice. Once with real error and once with "Queue overflow" error. Only the "Queue overflow" is actually stored in the queue.

commit a731cab4 (2015-12-05): Refactor scpi_init and resolve #66
closed 2016-02-13 by j123b567