j123b567/scpi-parser #68: need to be Thread Safe

pseudonym123, 2015-11-26:
I think it's much better if the parser running thread-safe.
One of the barriers I think is the error fifo.
Is there any plan to enhance the features?

j123b567, 2015-11-26:
There are two related issues #63 and #66.

I have currently no time to do so big change so there (#63) are some hints how to overcome this issue.

If you can provide some hints or pull request with some implementation, I will be happy.

Bug #66 is planned to be solved in near future.

pseudonym123, 2015-11-26:
If the user guarantee the context being thread-safe, I think the thread-safe could be accomplished by making the context having their own error fifo.

Do you think so?

j123b567, 2015-11-26:
Yes, see #66

j123b567, 2015-12-05:
Issue #66 is solved so this issue is also partially solved. I have created new issue focused only to locking.

closed 2015-12-05 by j123b567