j123b567/scpi-parser #70: Using scpi-parser in Sigrok?

timofonic, 2015-12-08:
As you see, there's some incomplete support of SCPI in sigrok.

http://sigrok.org/wiki/IEEE-488
http://sigrok.org/gitweb/?p=libsigrok.git;a=history;f=src/scpi;h=1a4ae2cf64210388b48e24b0bf8bf6093c028c1c;hb=HEAD

Can scpi-parser and sigrok projects collaborate in some way and share code? I think it would be better for both users and developers

j123b567, 2015-12-08:
We can cooperate, but currently I don't know how. You are developing master side and I'm developing slave (device) side. My license (2-clause BSD) is completely compatible with GPL so there is no problem to incorporate parts of this library to your code.

Do you have some idea, how we can cooperate? We can reuse some code and some know how.

j123b567, 2016-02-13:
No response for few months, so closing this issue. Feel free to reopen this or open new issue, if you have some nice ideas what to do.

closed 2016-02-13 by j123b567