j123b567/scpi-parser #73: Device-dependent error information

sandsbl, 2016-01-07:
New to this library. From the SCPI-99 documentation, the standard response to the query `SYSTem:ERRor[:NEXT]?` should be of the form `<Error/Event number>,"<Error/Event description>;<Device-dependent info>"`. Is there a defined way to append the `<Device-dependent info>` to the existing error description strings provided in `error.h` without overriding the `SCPI_SystemErrorNextQ` function provided in `minimal.c`?

Great library, thanks for the effort..

j123b567, 2016-01-07:
It is not possible without modifications.

I don't like big pre-allocated buffers and malloc/free and this feature leads into one of these. This means, that I should implement some configuration directive to enable malloc/free dependent functionality. Without this, functionality like this will not exist.

If you can implement it like I describe in following lines or better way, please do it and send me a pull request. Otherwise, I will implement this, but it will take long time as I'm permanently busy.

My idea is to:
 - Implement another function - e.g. `SCPI_ErrorPushEx` with additional string parameter (`const char*`).
 - Modify queue to hold also pointer to string
 - `SCPI_ErrorPushEx` will do "strdup" and use new string pointer for this queue
 - `SCPI_ErrorPop` will be extended to return also device dependent info. It can copy the stored string to prepared buffer and call `free` on the string (prefered), or it can expose pointer to the stored string and user of this function must free it manualy (not prefered).
 - Extend `SCPI_SystemErrorNextQ` to benefit from this change.

New function prototypes proposal:

```c
/* error.h */

/**
 * Push error with device dependendt information
 * @param context SCPI context
 * @param err SCPI error code
 * @param info Device dependent info or NULL
 */
void SCPI_ErrorPushEx(scpi_t * context, int16_t err, const char * info);

/**
 * Pop error from queue
 * @param context SCPI context
 * @param info Pointer to preallocated buffer
 * @param infolen Size of preallocated buffer info
 * @param oinfolen Return number of characters copied to info buffer
 * @return 
 */
int16_t SCPI_ErrorPop(scpi_t * context, char * info, size_t infolen, size_t * oinfolen);

/* types.h */
struct _scpi_fifo_item_t {
    int16_t value;
    void * ptr;
};
typedef struct _scpi_fifo_item_t scpi_fifo_item_t;

struct _scpi_fifo_t {
    int16_t wr;
    int16_t rd;
    int16_t size;
    scpi_fifo_item_t * data;
};
typedef struct _scpi_fifo_t scpi_fifo_t;

/* fifo_private.h */
void fifo_init(scpi_fifo_t * fifo, scpi_fifo_item_t * data, int16_t size) LOCAL;
scpi_bool_t fifo_add(scpi_fifo_t * fifo, int16_t value, const void * ptr) LOCAL;
scpi_bool_t fifo_remove(scpi_fifo_t * fifo, int16_t * value, const void ** ptr) LOCAL;
scpi_bool_t fifo_remove_last(scpi_fifo_t * fifo, int16_t * value, const void ** ptr) LOCAL;
```

cdwork, 2016-02-20:
1) In implementing the `void fifo_init (scpi_fifo_t * fifo, scpi_fifo_item_t * data, int16_t size) LOCAL;` will also need to change the `SCPI_Init`,` SCPI_ErrorInit` (if I understand correctly).
2) Perhaps instead

> - `SCPI_ErrorPop` will be extended to return also device dependent info. It can copy the stored string to prepared buffer and call `free` on the string (prefered), or it can expose pointer to the stored string and user of this function must free it manualy (not prefered) .

do on the basis of the method `size_t SCPI_ResultText (scpi_t * context, const char * data)` in the `parser.c` a new method, for example, `size_t SCPI_ResultTextCompl (scpi_t * context, const char * data [], size_t len)`. Where is the text will be displayed in the `writeData` array of lines `data`, separated by '`;`'. In this way we will avoid additional allocation buffer to prepare lines of constant line `<Error/event_description>` and already once copied line `<Device-dependent_info>`.

j123b567, 2016-02-20:
1) Of course. After these changes, we must correct all other functions using this.

2)
In "copy" case, `SCPI_ErrorPop` will copy the string to prepared buffer and free internal memory itself. We can still use `SCPI_ResultTextCompl`, but we need intermediate buffer for extended info. It is safe but it waste memory.

In your case, `SCPI_ErrorPop` will expose stored pointer to the string. `SCPI_SystemErrorNextQ` will print this extended information by `SCPI_ResultTextCompl` and after that, it must free the buffer itself. Problem is that in some implementations we can use preallocated buffer, that we don't want to free. It is generally not safe.

We can introduce some hibrid case and implement some `SCPI_ErrorCleanup` function to free popped element. It will do all memory manipulation in `SCPI_Error` subsystem and it will not need additional buffer. This should be our case.

```c
struct _scpi_error_t {
    int16_t error_code;
    const char * device_dependent_info;
}
typedef struct _scpi_error_t scpi_error_t;

/* pop element to provided error variable */
scpi_bool_t SCPI_ErrorPopEx(scpi_t * context, scpi_error_t * error);
/* cleanup all allocated buffers in "error" varable, but not the variable itself */
void SCPI_ErrorCleanup(scpi_t * context, scpi_error_t * error);
```

We should not forget, that the maximum string length of "Error/event_description" plus "Device-dependent_info" is 255 characters. We must limit this somewhere.

cdwork, 2016-02-20:
> Problem is that in some implementations we can use preallocated buffer, that we don't want to free.

I guess I'm missing something or not understand...
When we can use preallocated buffer?
Duplicate string of "user" `info` will be created in the method `SCPI_ErrorPushEx (context, err, info)`, further we will use "our" string with "our" pointer, and the string "user" can be anything you want. And when extracting in `SCPI_SystemErrorNextQ` we will work with "our" string, which, of course, `free` then complete.
I do not criticize. I'm trying to understand.

j123b567, 2016-02-20:
I mean alternate implementation of `SCPI_ErrorXYZ` where no malloc/free is used - e.g. if we can't use these functions at all.

cdwork, 2016-02-20:
Now I understand. I agree.
In this case, a circular buffer approach is possible. Make method works on the same principle as that of `malloc/free`: returning a pointer to this buffer or `NULL` if the size of the next line info exceeds the available space. Override these functions `config.h` same as `SCPIDEFINE_strnlen`, `SCPIDEFINE_strncasecmp` and etc.
I would start to do something, when we define the basics.

j123b567, 2016-02-20:
> I would start to do something, when we define the basics.

It will be very nice!

We can start with three possibilities available in `config.h`
 1. no support for Device-dependent error information at all (current state)
 2. variant with Device-dependent error information with malloc/free
 3. variant with Device-dependent error information but without malloc/free.

We should implement only 2. or 3. but not both for now. Please choose variant, which better suits your needs. I think, that 2. is easier for implementing the concept. 3. can be implemented as extension of 2.

First step is to extend the fifo.

Next step is to implement `SCPI_ErrorPushEx`, `SCPI_ErrorPopEx` and `SCPI_ErrorCleanup`. You can propose better names and prototypes for these functions. Proposals in this issue are not "the only one correct", they are here just to show the concept. Now I think, that we both know what the other mean.

Last step is to implement `SCPI_ResultTextCompl`/`SCPI_ResultTextMultiple`/`SCPI_ResultTextList` and extend `SCPI_SystemErrorNextQ` to print correct message.

Implementing/extending/correcting also unit tests for all these modifications will be nice.

j123b567, 2016-04-24:
Extended with tests and merged with some modifications.

closed 2016-04-24 by j123b567