j123b567/scpi-parser #75: XC32 compatibility

Toussaic, 2016-04-04:
Hi there,

For a full compatibility with Microchip XC32 compiler, I had to modify those things :

- the struct _scpi_interface_t, field "control" (reserved keyword), by "ctrl", used in ieee488.c writeControl function

- the struct _scpi_t, added const qualifier to the field "interface"

- the struct error_reg in error.c, field "bit" (reserved keyword), by "mask" and used in function SCPI_ErrorPush

Because I use an old version of the library, I can't give you the lines nor some other modifications I made, specially some private functions declarations I moved in "***_private.h" files to avoid multiple declarations when compiling.

And I had to modify the old function SCPI_ParamText in parser.c to skip the good number of bytes when using " " quoted text (look at the first char of the parameter and skip length+2 if '"')

Maybe one day I will try the new version of the library and give you some other modifications.

Toussaic, 2016-04-04:
I've just see the SCPI_ParamText error has already been issued.

j123b567, 2016-04-17:
I have made small fix to configuration and remove potentialy reserved bit keyword.

Please, before reporting problem, try current version. Older versions are not supported anymore. Many problems are already fixed and it take me lot of time to verify this.

Problem is that I can't reproduce your problems. I have created project in MPLABX with compiler C32 2.0.2 and XC32 1.40 and both compiles fine. There is no keyword "bit" nor "control" in clean XC32 project.

Can you tell me, what is your command line of compiler and what system header files are you including?

Toussaic, 2016-04-20:
I've checked my project and even if those words are recognized as keyword by the IDE, the project compile without warnings or errors. I asked for a explanation on the microchip forum and I'm waiting for an answer (if there is one).

I'm working with an old version because it is part of a bigger project I started two years ago and I don't want to change everything since it is in production on various devices. Sorry for reporting so late :)

I will try to upgrade to the new version on the future devices.

j123b567, 2016-04-20:
Bit can be a keyword on 8bit platform.

I understand the production state. If you are still using version 1.x there are many improvements in the code in version 2.x to be more SCPI compliant. It has support for handling arbitrary blocks, channel and numeric list and there are lot of small bug fixes. If you are interested in these features, try the new version.

Toussaic, 2016-04-21:
Both "bit" and "control" are reserved keywords in XC8. "Bit" is a 1 bit length integer data type and "control" qualifies special registers outside memory space used by some 8 bit microcontrollers.

(And I remember now that I started my first prototype on a 8 bit microcontroller, this is why I needed to change those field names ...)

j123b567, 2016-04-24:
There are no more issues wih XC32 so closing this.

closed 2016-04-24 by j123b567