j123b567/scpi-parser #77: Question: What are command tags?

llahti, 2016-05-30:
So far i have been trying to search information what are the command tags from the documentation page http://j123b567.github.io/scpi-parser/. But i didn't find any useful information about what are those and how those should be used.

Actually reason why i am asking is that when i'm trying to compile scpilib i will get plenty of compiler warnings caused by missing command tag definition. I'm using **arm-none-linux-gnueabi-g++** **(ctng-1.6.1)** **4.4.3** to compile this project.

Because of the missing tag definition i will get following warning
> warning: missing initializer for member '_scpi_command_t::tag'

I did small search in source code and find that the command tags seems to be optional. 
`struct _scpi_command_t {
        const char * pattern;
        scpi_command_callback_t callback;
#if USE_COMMAND_TAGS
        int32_t tag;
#endif /* USE_COMMAND_TAGS */
    };`

For reference I'm constructing command list by following way.
`const scpi_command_t scpi_commands[] = {
    { "*CLS", SCPI_CoreCls,},
    { "*ESE", SCPI_CoreEse,},`

j123b567, 2016-05-30:
This functionality is here for sharing one callback with multiple commands. You can later detect real command in callback by its tag using [SCPI_CmdTag](http://j123b567.github.io/scpi-parser/api/SCPI_CmdTag/)

Example:
```c
....
{ "CMDabcd", CMDabcd_cb, 0},
{ "CMDabcd?", CMDabcd_cb, 1},
....

scpi_result_t CMDabcd_cb(scpi_t *context) {
  if (SCPI_CmdTag(context) == 1) {
     // command is query
  } else {
     // it is standard command
  }
}
```

This is just a problem using c++ compiler. Using c++ compiler is not well tested so there may by lot of warnings.

You can disable this functionality by defining `-DUSE_COMMAND_TAGS=0` in gcc/g++ command line for both - the library and user code. Or alternatively define `-DSCPI_USER_CONFIG` and use `scpi_user_config.h` for more configurations.

You can define your commands by 

```c
const scpi_command_t scpi_commands[] = {
 { "*CLS", SCPI_CoreCls, 0},
 { "*ESE", SCPI_CoreEse, 0},
...
```

to suppress this warning in g++.

llahti, 2016-05-30:
This answers my questions very well :) Now command definitions work without warnings. Thanks.

You're right that G++ compiler produces quite a lot of warnings. I tested it with G++ 4.4.3 and  5.3.1.
Regardless of warnings code is compiled successfully with G++ 5.3.1, but unfortunately our cross-compiling toolchain with G++ 4.4.3 produces errors because of `isfinite(x)` macro. I can provide more information about it and possible fixes tomorrow.

closed 2016-05-31 by j123b567