j123b567/scpi-parser #78: utils.c:  SCPI_dtostre() and isfinite() macro issues with G++ 4.4.3

llahti, 2016-05-31:
This issue is most likely specific to G++ 4.4.3 and/to codesourcery crosscompiler toolchain for arm-linux.

Anyhow compiler throws error because of undefined reference to 'isfinite' macro.

```
utils.o: In function 'SCPI_dtostre':
utils.c:(.text+0x1080): undefined reference to 'isfinite'
utils.c:(.text+0x1110): undefined reference to 'isfinite'
```

I have temporarily fixed the issue by using **finite()** function for G++ 4.4.3 which is about the same than **isinfinite()** macro. I wonder if there would be more correct way to fix this issue?
For reference.. Below link describes libc floating point functions.
https://www.gnu.org/software/libc/manual/html_node/Floating-Point-Classes.html

```c
    // Modification to original by Lauri Lahti 2016.05.30
    // G++ 4.4.3 have problems with isfinite(x) macro
    // Replaced by finite(x) which is almost the same.
    //
#if (__GNUC__ == 4) && (__GNUC_MINOR__ == 4) && (__GNUC_PATCHLEVEL__ == 3)
    if (!finite(__val)) {
#else
    if (!isfinite(__val)) {
#endif
        if (isnan(__val)) {
            strcpy(s, (__flags & SCPI_DTOSTRE_UPPERCASE) ? "NAN" : "nan");
        } else {
            strcpy(s, (__flags & SCPI_DTOSTRE_UPPERCASE) ? "INF" : "inf");
        }
        strncpy(__s, buffer, __ssize);
        return __s;
    }
```

j123b567, 2016-05-31:
Can you please post output of `g++ -dM -E - < /dev/null` here or to pastebin or so.

llahti, 2016-05-31:
I'm curious is there some specific \#define you're looking for?

Command `arm-none-linux-gnueabi-g++ -dM -E - < /dev/null` gives following output
```c
#define __DBL_MIN_EXP__ (-1021)
#define __FLT_MIN__ 1.17549435e-38F
#define __CHAR_BIT__ 8
#define __WCHAR_MAX__ 4294967295U
#define __DBL_DENORM_MIN__ 4.9406564584124654e-324
#define __FLT_EVAL_METHOD__ 0
#define __unix__ 1
#define __DBL_MIN_10_EXP__ (-307)
#define __FINITE_MATH_ONLY__ 0
#define __ARMEL__ 1
#define __GNUC_PATCHLEVEL__ 3
#define __DEC64_MAX_EXP__ 385
#define __SHRT_MAX__ 32767
#define __LDBL_MAX__ 1.7976931348623157e+308L
#define __GXX_TYPEINFO_EQUALITY_INLINE 0
#define __UINTMAX_TYPE__ long long unsigned int
#define __linux 1
#define __DEC32_EPSILON__ 1E-6DF
#define __CHAR_UNSIGNED__ 1
#define __LDBL_MAX_EXP__ 1024
#define __linux__ 1
#define __SCHAR_MAX__ 127
#define __DBL_DIG__ 15
#define __SIZEOF_INT__ 4
#define __SIZEOF_POINTER__ 4
#define __USER_LABEL_PREFIX__ 
#define __STDC_HOSTED__ 1
#define __LDBL_HAS_INFINITY__ 1
#define __FLT_EPSILON__ 1.19209290e-7F
#define __APCS_32__ 1
#define __LDBL_MIN__ 2.2250738585072014e-308L
#define __DEC32_MAX__ 9.999999E96DF
#define __SIZEOF_LONG__ 4
#define __DECIMAL_DIG__ 17
#define __gnu_linux__ 1
#define __LDBL_HAS_QUIET_NAN__ 1
#define __GNUC__ 4
#define __FLT_HAS_DENORM__ 1
#define __SIZEOF_LONG_DOUBLE__ 8
#define __BIGGEST_ALIGNMENT__ 8
#define __DBL_MAX__ 1.7976931348623157e+308
#define __DBL_HAS_INFINITY__ 1
#define __DEC32_MIN_EXP__ (-94)
#define __THUMB_INTERWORK__ 1
#define __LDBL_HAS_DENORM__ 1
#define __DEC128_MAX__ 9.999999999999999999999999999999999E6144DL
#define __DEC32_MIN__ 1E-95DF
#define __DBL_MAX_EXP__ 1024
#define __DEC128_EPSILON__ 1E-33DL
#define __LONG_LONG_MAX__ 9223372036854775807LL
#define __SIZEOF_SIZE_T__ 4
#define __SIZEOF_WINT_T__ 4
#define __GXX_ABI_VERSION 1002
#define __SOFTFP__ 1
#define __FLT_MIN_EXP__ (-125)
#define __DBL_MIN__ 2.2250738585072014e-308
#define __DEC128_MIN__ 1E-6143DL
#define __REGISTER_PREFIX__ 
#define __DBL_HAS_DENORM__ 1
#define __NO_INLINE__ 1
#define __FLT_MANT_DIG__ 24
#define __VERSION__ "4.4.3"
#define __DEC64_EPSILON__ 1E-15DD
#define __DEC128_MIN_EXP__ (-6142)
#define unix 1
#define __SIZE_TYPE__ unsigned int
#define __ELF__ 1
#define __FLT_RADIX__ 2
#define __LDBL_EPSILON__ 2.2204460492503131e-16L
#define __VFP_FP__ 1
#define __SIZEOF_PTRDIFF_T__ 4
#define __DEC32_SUBNORMAL_MIN__ 0.000001E-95DF
#define __FLT_HAS_QUIET_NAN__ 1
#define __FLT_MAX_10_EXP__ 38
#define __LONG_MAX__ 2147483647L
#define __DEC128_SUBNORMAL_MIN__ 0.000000000000000000000000000000001E-6143DL
#define __FLT_HAS_INFINITY__ 1
#define __unix 1
#define __DEC64_MAX__ 9.999999999999999E384DD
#define __CHAR16_TYPE__ short unsigned int
#define __DEC64_MANT_DIG__ 16
#define __DEC32_MAX_EXP__ 97
#define linux 1
#define __LDBL_MANT_DIG__ 53
#define __DBL_HAS_QUIET_NAN__ 1
#define __WCHAR_TYPE__ unsigned int
#define __SIZEOF_FLOAT__ 4
#define __DEC64_MIN_EXP__ (-382)
#define __FLT_DIG__ 6
#define __INT_MAX__ 2147483647
#define __FLT_MAX_EXP__ 128
#define __DBL_MANT_DIG__ 53
#define __DEC64_MIN__ 1E-383DD
#define __WINT_TYPE__ unsigned int
#define __SIZEOF_SHORT__ 2
#define __LDBL_MIN_EXP__ (-1021)
#define __arm__ 1
#define __LDBL_MAX_10_EXP__ 308
#define __DBL_EPSILON__ 2.2204460492503131e-16
#define __SIZEOF_WCHAR_T__ 4
#define __DEC_EVAL_METHOD__ 2
#define __INTMAX_MAX__ 9223372036854775807LL
#define __FLT_DENORM_MIN__ 1.40129846e-45F
#define __CHAR32_TYPE__ unsigned int
#define __FLT_MAX__ 3.40282347e+38F
#define __SIZEOF_DOUBLE__ 8
#define __FLT_MIN_10_EXP__ (-37)
#define __INTMAX_TYPE__ long long int
#define __DEC128_MAX_EXP__ 6145
#define __GNUC_MINOR__ 4
#define __DEC32_MANT_DIG__ 7
#define __DBL_MAX_10_EXP__ 308
#define __LDBL_DENORM_MIN__ 4.9406564584124654e-324L
#define __STDC__ 1
#define __ARM_ARCH_4T__ 1
#define __PTRDIFF_TYPE__ int
#define __DEC64_SUBNORMAL_MIN__ 0.000000000000001E-383DD
#define __ARM_EABI__ 1
#define __DEC128_MANT_DIG__ 34
#define __LDBL_MIN_10_EXP__ (-307)
#define __SIZEOF_LONG_LONG__ 8
#define __LDBL_DIG__ 15
#define __GNUC_GNU_INLINE__ 1
```

j123b567, 2016-05-31:
It may not be a problem of all G++ 4.4.3 but only of some specific ones. So just need to know.

j123b567, 2016-05-31:
Can you please try latest version?

llahti, 2016-06-01:
Latest version works well with `g++ (Ubuntu 5.3.1-14ubuntu2.1) 5.3.1 20160413`

`arm-none-linux-gnueabi-g++ (ctng-1.6.1) 4.4.3` have the same old issue with isfinite(). I wonder whether i am missing something in my cross-compiling tool chain?

```
utils.o: In function `SCPI_dtostre':
utils.c:(.text+0x1164): undefined reference to `isfinite'
utils.c:(.text+0x1208): undefined reference to `isfinite'
```

llahti, 2016-06-01:
This new way of defining whether to use isfinite() or finite() seems to be good addition.
By using this specific workaround in `cc.h` i can compile code successfully by using `arm-none-linux-gnueabi-g++ (ctng-1.6.1) 4.4.3`

Following code is inserted on line 118 in cc.h
```c
/*GCC 4.4.3 arm-none-gnueabi-g++ */
#if (__GNUC__ == 4) && (__GNUC_MINOR__ == 4) && (__GNUC_PATCHLEVEL__ == 3)
    #undef HAVE_ISFINITE
    #undef HAVE_FINITE
    #define HAVE_ISFINITE 0
    #define HAVE_FINITE 1
#endif
```

Well.. i feel these compiler specific fixes may not be the best solution. Maybe there are some better ways as well.

j123b567, 2016-06-01:
What is your C library - glibc, uclibc, newlib, ...? And what is its version?

j123b567, 2016-06-01:
I still can't reproduce this. I have compiled arm toolchain with GCC 4.4.7, glibc-2.20 and everything works with gcc or g++ compiler. `finite` is BSD function so are you using Android's Bionic C library?

j123b567, 2016-06-01:
You can now compile your code with `-DHAVE_FINITE=1` option without any change to sources.

llahti, 2016-06-03:
I didn't realize i could just configure those defines from command line. Always can learn something new.

Then what comes to C-library i try to find out more information about that. That i know for sure that it's not android bionic.

I'm more and more convinced this is a problem with our cross-compiling system. So at the moment i feel there is no need to put more effort on this.

j123b567, 2016-06-04:
> I didn't realize i could just configure those defines from command line. Always can learn something new.
It was not possible in previsous versions.

> I'm more and more convinced this is a problem with our cross-compiling system. So at the moment i feel there is no need to put more effort on this.

Now, it is much more flexible to define availabble features but without autoconf/cmake/.. build script I can't do much. So closing this issue.

closed 2016-06-04 by j123b567