j123b567/scpi-parser #81: SCPI-99:9.4 Implement "fan-out" register support

sandsbl, 2016-09-26:
As part of our SCPI instrument development, we have come upon a need to include "fan-out" registers for certain bits in QUEStionable and OPERation status registers as defined in section 9.4 of the SCPI-99 spec. I propose adding this as a feature-request for the scpi-parser library. I am willing to work on this feature, as we are in the most immediate need for it, but I would like to get some feedback on my thoughts on how to proceed. I am thinking to take the following steps to implement this feature:

+ **Add the member `scpi_fanout_reg_t * fanout_registers;` to `_scpi_t` context structure**  
This would be a pointer to an array of device-specific fanout registers that would need to be defined in `scpi_context.c`. Each fanout register could be defined in `types.h` as
    ```cpp
struct  _scpi_fanout_reg_t {
    scpi_reg_val_t val;
    scpi_reg_name_t parent_reg;
    scpi_reg_val_t parent_bit;
    int32_t parent_fanout_reg;
    scpi_bool_t is_nested;
};
typedef struct _scpi_fanout_reg_t scpi_fanout_reg_t;
```
This specifies the register value `val`, its standard parent register `parent_reg` (one of OPER or QUES), and the summary bit to which this fanout is logically summed `parent_bit`. Since the standard specifies the potential for multiple layers of nesting fanout registers, the last two members would handle this case. If `is_nested` is `TRUE`, then the fanout register at index `parent_fanout_reg` is updated as a summary bit. 

+ **Extend `SCPI_RegXXX` functions in `IEEE488.c`**  
The following functions would need to have their "fanout" analogs added to handle `scpi_fanout_reg_t` registers:  
`SCPI_RegSet` -> `SCPI_RegSetFanout`
`SCPI_RegSetBits` -> `SCPI_RegSetFanoutBits`
`SCPI_RegClearBits` -> `SCPI_RegClearFanoutBits`
`SCPI_RegGetFanout` -> `SCPI_RegGetFanout`
`regUpdate` -> `regUpdateFromFanout`  
Additionally, there is a need for a new function `SCPI_RegCheckBits` that would check if a particular register bit from OPER or QUES is a summary bit. This would presumably be used e.g. when a **\*CLS** command is issued or when a summary bit is read in order to clear its associated fanout registers.  

+ **Update `SCPI_CoreCLS` in `IEEE488.c`**  
Clear fanout registers.

+ **Update QUEStionable and OPERation status callbacks as well as `SCPI_StatusPreset` in `minimal.c`**  
Handle fanout registers.

Please advise with suggestions or issues I may have forgot.

j123b567, 2016-09-26:
I thing, there was no discussion about this in the past.

Can you propose a pull request with this functionality?

sandsbl, 2016-09-27:
Yes, I can start working on it.

PolyPik, 2020-04-15:
@sandsbl Have you implemented this feature?

sandsbl, 2020-04-16:
No, I haven't had a chance to, sorry. Moved to another project.

PolyPik, 2020-04-16:
Well you did a pretty good job of describing what needs to be done. I think I can implement the changes on my own.

j123b567, 2023-01-16:
Should be fixed by #114

closed 2023-01-16 by j123b567

jancumps, 2023-09-13:
@PolyPik , do you happen to have an example or test case that uses device specific registers?
I'm trying to use the functionality, to flag that a digital input has changed level ([blog](https://community.element14.com/technologies/test-and-measurement/b/blog/posts/pst-experimental-event-trigger-support-for-pico-scpi-labtool)). I may be able to find out how to use the extension. An example or reference project would be helpful.

jancumps, 2023-09-22:
I figured it out. I managed to build a set of instrument specific registers,  
make them fan out to a  bit of the OPERATION STATUS register, etc...  
Took me a good week to fully make it happen :) - partly because I wanted a good integration with TinyUSB's TMCUSB with service requests. 

documentation [link](https://community.element14.com/technologies/test-and-measurement/b/blog/posts/pst-experimental-event-trigger-support-for-pico-scpi-labtool).