j123b567/scpi-parser #88: multiple parameters

calmacalma, 2017-07-28:
Hi , i'm trying to define scpi commands that take multiple hexadecimal parameters:
how should i implement this (which structure should i use?)

j123b567, 2017-07-29:
Hexadecimal number is just number, so any function that reads numeric parameter will work. `SCPI_ParamUInt32` or `SCPI_ParamUInt64` are prefered for this. Example of reading two double parameters is here https://github.com/j123b567/scpi-parser/blob/master/examples/common/scpi-def.c#L99 First parameter is mandatory, second is not in this example. Reading UInt32 is similar, just use correct data type and different name of SCPI_ParamXYZ function.

If you need to read large number of parameters, there are functions SCPI_ParamArrayXYZ like `SCPI_ParamArrayUInt32`.

    static scpi_result_t SomeCallbackMultipleParameters(scpi_t * context) {
        size_t o_count;
        uint32_t data[32]; /* can handle up to 32 parameters */

        if(!SCPI_ParamArrayUInt32(context, data, 32, &o_count, SCPI_FORMAT_ASCII, TRUE)) {
            return SCPI_RES_ERR;
        }

        /* @fixme: do something with data. Only o_count values are valid. */
        return SCPI_RES_OK;
    }

In this example, up to 32 parameters will be read. If the mandatory flag is TRUE, at leas one number must be present. If you need exactly 32 parameters, just return SCPI_RES_ERR if o_count is not 32.

calmacalma, 2017-08-06:
Hi, thank you very much for your response :) but i still have one question : when i define the scpi command that will read 4 numbers for example and i want to enter these numbers in hex format e.g:
COmmand#Hnum1 #Hnum2 #Hnum3 #Hnum4 i always get an error for the scpi command format.. any suggestion to define such command??
thanks!!!

j123b567, 2017-08-06:
Please tell me exact command line you try to use. Tell me also exact number of error you get.
I thing, that you are missing space between command header and parameters. This space is always needed.

Example:
If you have definition of command pattern e.g.

    {"COmmand", SomeCallbackMultipleParameters}

Here are examples of calling this command

    co #H920 #H180 #H3F0 #H1F0
    command #H920 #H180 #H3F0 #H1F0
    CoMmAnD #H920 125 1024 #H1F0
    command 1 2 3 4

Here are examles of wrong calling this command

    command#H920 #H180 #H3F0 #H1F0

calmacalma, 2017-08-06:
when i write : command #H920 #H180 #H3F0 #H1F0 for example i get these errors:
**ERROR -101, "invalid character"
**ERROR -101, "invalid character"
**ERROR -113, "undefined header"

cdwork, 2017-08-06:
You forgot commas...
command #H920, #H180, #H3F0, #H1F0

j123b567, 2017-08-06:
You are right! Its a long time I have done something with SCPI. I have corrected examples.

calmacalma, 2017-08-06:
now it works, thanks a lot :-)

closed 2017-08-06 by j123b567