j123b567/scpi-parser #93: error: undefined reference to 'frexp', 'modf'

jfichtner, 2017-08-14:
When I try to build the project (from current master branch) I get errors in multiple test projects related to the math.h functions frexp and modf. Adding "-lm" (linking libm.so) to the CFLAGS variable in each of the respective makefiles fixes the issue. I'm not familiar enough with GNU make to know if this fix is appropriate for a pull request or not, or if perhaps this could have other side effects on other build environments / targets. Please advise.

j123b567, 2017-08-14:
What compiler are you using? (and what build environment).

jfichtner, 2017-08-14:
I'm using arm-linux-androideabi-gcc (GCC) 4.8 on Ubuntu 14.04, targeting android.

j123b567, 2017-08-16:
It seems to be an "issue" of Bionic C library. Anyway, these functions are from libm and it should be linked by '-lm'. I will try it with current build on linux.

j123b567, 2017-08-16:
Can you try latest version if it fixes all problems for you?

jfichtner, 2017-08-18:
Yes, that fixes my build issues, thanks!

closed 2017-08-18 by jfichtner