j123b567/scpi-parser #95: Large arbitrary block of data

mvladic, 2017-11-15:
Hi!

I would like to transfer a large arbitrary block data to the device, chunk by chunk, since it can't fit all at once inside device RAM memory. Currently, I think, SCPI_ParamArbitraryBlock expects whole block to be inside the memory. Is there any other strategy to do this? 

For the other way around, when I need to transfer large arbitrary block from device, I can use SCPI_ResultArbitraryBlockHeader and then SCPI_ResultArbitraryBlockData multiple times and it works fantastic!

For the context, let me tell you that I transfer data from PC to the SD card on the device, and vice versa.

mvladic, 2017-11-15:
I just figured out one strategy to do this. I will send one chunk of data with one SCPI command, i.e. each chunk will be different SCPI command.

For example, to transfer file from PC to device, I will do the following:
```
MMEMory:DOWNload:FNAMe "/lists/sin.csv"
MMEMory:DOWNload:DATA #264.... // send 1st 64 bytes
MMEMory:DOWNload:DATA #264.... // send 2nd 64 bytes
...
```

cdwork, 2017-11-15:
Hi!
See this post: https://github.com/j123b567/scpi-parser/issues/80

j123b567, 2017-11-15:
This is good workaround.

Current library design dos not support messages larger then the buffer. This is also the reason, I can't support indefinite format of arbitrary data, see issue #24.

Another approach would be, to add another API to the library. User will define user handler just for 

 - detection of command header
 - detection of parameter, it can be called multiple times for one ARB parameter
 - detection of end of command

Detection of correct command will be on responsibility of the user of the library.

This is how another parsers works e.g. https://github.com/nodejs/http-parser

mvladic, 2017-11-15:
I think I can live with the workaround. Thanks!