j123b567/scpi-parser #99: Using gperf to search into tables ?

CdeMills, 2018-02-26:
Hello,

I compiled and run libscpi on a PSOC5 platform. It works ... but slowly.

A lot of search on tables are performing using linear walk, testing items one at a time. Would it be possible to use gperf to generate optimized code where each search is about extracting one key from an array and looking one entry in a table. The item match or not.

This implies:
1) split the code so that each table resides in its own file in order to process it with gperf
2) run gperf :-)
3) integrate only the processed table in the main executable
added bonus: each table is defined as static and const, and may reside in program space. Embedded systems tends to have a small amount of RAM and a greater amount of FLASH for the code.

I'm ready to start hacking on this. How should I interact with this project ?

Regards

Pascal

j123b567, 2018-02-26:
Hello Pascal,

in general, all ideas are welcome.

Searching `scpi_commands[]` table is slow, because it is not just string comparision on equality. IMHO it can't be speed up by any general purpose hashing or sorting like `gperf`. Imagine real world pattern `[:SOURce]:CURRent[:LEVel][:IMMediate][:AMPLitude]`, this should match command `sour:curr:lev:imm:ampl`, `curr`, `:current:aplitude`, etc. - there are in total `2*3*2*3*3*3=324` different commands matching this pattern! Assuming case insensitive matching.
There are other patterns, eg. `OUTPut#[:STATe]` can match `outp1`, `:output2`, `outp:state`, etc. This means `2*2*N*3=12*N` variants.

However, there are still some ways, how this can be speed up. You can propose an interface to define own searching function. I will give you `scpi_context` and command name and you will give me command callback function pointer or something like that.

Later, you can play freely with the optimization.

There are at least these ways, how can you improve this

1. Using lexer generators - but you must use regular expressions to define all command patterns. You can play with re2c, (f)lex, (b)yacc, antlr, or any other tool.
2. Using tree structure instead of linear table.
3. Using gperf, because you have some better idea and I'm not true

If you would like to improve lexer and parser of commands itself, there is possibility to rewrite some parts in `re2c` or so. It can save some time.

Jan

CdeMills, 2018-02-26:
Hello Jan,

thank you for answering me.

The basic problem is rather simple: we have two kinds of commands. Those
without any fuss ('*CLS'), and those which requires special handling, with
optional parts and so on.

So I splitted the search in parser.c; function findCommandHeader:
1) do a fast search through  context->in_scpicmd_name(header, len) (new
function) which use code generated from gperf. The search space contains
all the short form strings
2) if not found, do the search using the original method, which implies to
handle all the optional parts

In the target program, scpi-def.c is split in three parts:
1) the definition of all the functions
2) code and table to search on short form strings
3) table with the long form strings
as 1) define static functions, 2) and 3) are merely added through an
'#include' statement

The result is that if you are using only short form strings, you get a
constant time access. The search on long form strings is faster too, as
this table is now shorter by the removal of all the '*XXX" commands.

I forked your tree and made changes up to the point where
examples/test-parser/ can be generated and tested; giving exactly the same
output as your version.

Could you have a look on https://github.com/CdeMills/scpi-parser ?

I'm not a specialist of Makefiles ... if examples/test-parser buid fails,
go to this directory and type:
make -B ../common/scpi-const.c
make test

The first command forces a rebuild of ../common/scpi-const.c though gperf

Regards

Pascal

j123b567, 2018-02-26:
Hello Pascal,

I saw your changes but for common instrument, this make sense just for common commands (commands starting with `*`). 

All other commands commonly have short and long form, so you choose some forms of commands, which will be fast and other forms of the same command will be slow. User must know which form will be fast so he will not be surprised.

I don't like putting almost same lines in two different files but it is price for performance.

I have another idea, that would be universal, easy to maintain and it will not consume so much space. Idea is not to use gperf, but generate some hash table on demand. The capacity of hash table will be N and N latest/most used commands will be there. First call of every command will be slow but all other calls of the same command will be fast. You can fill the table by commonly used commands at startup. You can choose N depending on your free RAM.

Jan

j123b567, 2018-03-04:
I have another idea with approach described in http://www.heeden.nl/statichashc.htm

It can be generated at compile time without external tool and duplicating of the code.
Disadvantage is that it is not perfect hash so the complexity of the search will be O(log(n)) and not O(1) and there are possibility with duplicates. It is still better then current linear search with string comparision which is somethink like O(n×m).

jancumps, 2018-03-04:
I'm a user, not a contributor, so feel free to ignore:

I like the suggestion of opening the API for other search methods more than choosing one now.
It allows to serve all without too much interference or additional dependencies. Once one option has proven to be good, we can adapt.
I'll help integrating it then.

j123b567, 2018-03-04:
I will create the API change soon with my proposal and with slightly changed your proposal. I think others can benefit from both.

CdeMills, 2018-03-04:
Hello 
I implemented the hash search on the PSOC5 (32 bits 80 MHz Cortex M3 platform). It helped, but not to the point where the delay become unnoticeable. This search is probably not the most heavy burden. Maybe it would be time to check with gprof ?

OTOH, I also had another idea. Instead of processing all at once and searching a long list, the analysis should be constructed along small lists. Each element would contain a pattern, a pointer to next pattern, and a pointer to function. The two latter would permit to decide if a pattern is final or not, and if a function could be launched if the input pattern is over.
The next step would be to construct a tree with all the possibilities to navigate. Basically we would duplicate the document explaining the commands: have a first level list, then a number of second levels lists, and so on. Once the tree is constructed, it should be compiled statically to a parsing engine.

This way the burden is performed once at compile time. 

What do you think of the idea ?

Pascal

jancumps, 2018-03-04:
Pascal, I like improving search delays in the library. I'd like that the exercise doesn't impact code size on small embedded processors. 
If a solution can be found that doesn't increase memory space and code space, and allows for improvements, I'd be all for it - if any dependencies added are healthy projects.

karlp, 2018-03-04:
"This search is probably not the most heavy burden"  well yeah, that's what I was thinking on the beginning.  When I've implemented this on a smaller MCU, it was a good chunk of _code_ but it had no performance impact worth talking about.  you're psoc5 should be more than capable of parsing scpi faster than you can deal with.  Are you sure you simply don't have delays with actually getting the scpi in and out of your system itself?  What were you using to measure things in the first place that led you to this desire to optimize the search paths?

j123b567, 2023-01-16:
Closing it as not a bug of this library.

closed 2023-01-16 by j123b567