j123b567/scpi-parser #105: Send multiple commands

martsalz, 2019-01-13:
First of all, thanks for this great library. 

How many SCPI commands can be processed in one processing step? How do I have to separate the commands from each other in the string?

Thank you very much

jancumps, 2019-01-13:
I haven't checked how long the command string can be, but you can send multiple commands using the standard SCPI ; separator (semicolon)

j123b567, 2019-01-13:
Number of commands is limited just by input buffer size. All commands must be in the buffer. Library will start to execute them after receiving the last endline character.

Command after ; is executed in the tree path of the previous command. Eg. 
`conf:volt:dc 1;ac 2` is valid and do the same as `conf:volt:dc 1` and `conf:volt:ac 2` executed separately. If you would like to run the next command in a root path, you should prefix it with : eg. `conf:volt:dc 1;:meas:volt:dc?`

closed 2019-06-19 by j123b567