j123b567/scpi-parser #106: Please help with library

alivecrim, 2019-03-04:
I absolutely do not understand how to use the library. Can you explain what exactly it does and for what? In the near future, I need to write software that will send SCPI commands for the equipment and receive answers. Can I use this library?
For example:
1. Open the port on the hardware
2. Send a command to the equipment
3. Get an answer.
In which of these stages can I use it?
All I understand is that inside it it breaks the line, it is defined by the command or by the query. But for what? Please see some kind of life example. Thank you very much. Sorry for my English, this is not my native language.

j123b567, 2019-03-04:
No, you can't.

This library is for implementing new instrument/equipment and to help implementing its own firmware.

To communicate with instrument from PC, you should use other libraries - e.g. NI VISA, sigrok or just do it yourself, it is not so hard.

Cheribat, 2019-03-19:
I've tried the cvi project on labwindow/cvi 2015，and here comes four errors in utils.c, like this:
Build Status (TestLibSCPI_GUI.prj - Debug)
 utils.c - 4 errors, 3 warnings
  323, 12    error: implicit declaration of function 'strtof' is invalid in C99. Make sure that you include the function prototype. 
  "config.h"(252,1)    note: expanded from macro 'SCPIDEFINE_strtof'
  965, 49    warning: implicit conversion loses integer precision: 'int' to 'char' 
  984, 30    warning: implicit conversion loses integer precision: 'int' to 'char' 
  1010, 16    error: implicit declaration of function 'signbit' is invalid in C99. Make sure that you include the function prototype. 
  "config.h"(238,1)    note: expanded from macro 'SCPIDEFINE_signbit'
  1017, 17    error: implicit declaration of function 'isnan' is invalid in C99. Make sure that you include the function prototype. 
  "config.h"(270,1)    note: expanded from macro 'SCPIDEFINE_isnan'
  1027, 10    error: implicit declaration of function 'isfinite' is invalid in C99. Make sure that you include the function prototype. 
  "config.h"(246,1)    note: expanded from macro 'SCPIDEFINE_isfinite'
  1112, 34    warning: implicit conversion loses integer precision: 'int' to 'uint16_t' (aka 'unsigned short') 
Build failed.
---
How can I  fix it?

j123b567, 2019-03-19:
You can try to put some defines to `scpi_user_config.h` file.

```
#undef HAVE_STRTOF
#undef HAVE_SIGNBIT
#undef HAVE_ISNAN
#undef HAVE_ISFINITE
```

I don't have LabWindows CVI so I have no posiibility to test it. In the past, they have outdated C compiller that don't obey C standards. Maybe, current state of CVI is much worser then in the past. It seems that it tells that it understad C99 standard but it don't.

closed 2019-06-19 by j123b567