j123b567/scpi-parser #107: Intent of library use and configuration options

jrmclaugh, 2019-06-18:
We're trying to add a list of our own error numbers and error texts, but can't seem to figure out the proper way to use the USE_USER_ERROR_LIST and USE_DEVICE_DEPENDENT_ERROR_INFORMATION.  We're porting this to an embedded system.

Some of this is probably just a normal C scoping and use issue, but just a couple questions as far as how you intended this library to be used:
- Are we supposed to edit the config.h, or should we be able to just #define options before including the first scpi.h?
- Is this more intended to be built as a static library, rather than including the source code directly?

j123b567, 2019-06-19:
Putting defines in the file before including scpi.h is not enough, because these defines are used just by your source and not by the library source files. Please study something about C compilation process so you will see the obvious problem of this. (You need to understand at least the preprocessing step http://cs-fundamentals.com/c-programming/how-to-compile-c-program-using-gcc.php#compile-c-program-preprocessing  Every single C file is preprocessed separately without knowing any other C file.)

The clean way 1:
If you have few values to configure, just put them to compiler directives for all C files  (-D option for gcc, many IDEs have ability to define compilers directives, etc. If you can specify your IDE and compiler, I can point you, where to find it)

The clean way 2:
You can define `SCPI_USER_CONFIG` as compiler directive and later, you can use `scpi_user_config.h` in any place you have configured as include search path.
Then put all your defines in this file.

The dirty way:
Modify directly config.h.
You can just remove `#ifdef/#endif` guard around inclusion of `scpi_user_config.h` if you don't know, what compiler directives are, so you can use `scpi_user_config.h` and it will work. But the better way is to use "The clean way 2" for this and learn the propper way.

The wrong way:
Please don't modify every single file to have some define in it. There are better ways.

jrmclaugh, 2019-06-19:
Thanks for the prompt response and that resource!  I do remember about compilation units now, I was getting mixed up with another IDE that uses very nonstandard C and just #defining before the include is enough (it actually didn't even use "include", but something similar).

That's what I suspected in terms of intended use (clean way 2, specifically), we had just been having strange issues even when defining at the project level.  Obviously the preferred method is to leave it alone and just configure as necessary!  Based on the chain of #include files and where we had the #defines, it looked like that compilation unit should have seen it, but it still wasn't.  Just wanted to make sure there was some other obvious thing we were missing.  Cleaning everything up and stepping back seems to have cleared it up.

Anyway, thanks for confirming the intended usage and for the very nice library!

closed 2019-06-19 by jrmclaugh