j123b567/scpi-parser #110: IP format input parameter

Thomas2908, 2020-03-05:
I would like to implement IP setup CMD such as SYST:ETH:IP 10.0.0.35
but I obtain error -101, "Invalid character". When I try this SYST:ETH:IP "10.0.0.35" it is OK.
It is necessary to enter the ip address as a string? Or is a better way to work with parameters of ip format?

Thank you for your answer.
Tomas.

j123b567, 2020-03-05:
IPv4 format is not defined in SCPI specification. It is always better to use strings as others do.

[Keysight](http://rfmw.em.keysight.com/wireless/helpfiles/n5106a/scpi_commands_syst_comm.htm):
```
:SYSTem:COMMunicate:LAN:ADDRess "<ip address>"
:SYSTem:COMMunicate:LAN:ADDRess?
```

SCPI-99 specification itself:
```
:SYSTem:COMMunicate:SOCKet1:ADDRess ”123.456.789.1"
```

Thomas2908, 2020-03-05:
Hi,

OK.
I saw this format in some commands manual. This manual claims to meet some IEEE488.2 & IEC 60488-2 standards. So I thought, that is standard access. I'm just getting acquainted with SCPI cmds. So I will use it as a string.

Thank you for your fast reply.
Tomas

j123b567, 2020-03-05:
This format is allowed only as a response by `IEEE488.2 chapter 8.7.11 <ARBITRARY ASCII RESPONSE DATA>` but it is not allowed as a command parameter.

Closest allowed parameter type is `chapter 8.7.11 <CHARACTER PROGRAM DATA>` which can be lowercase or uppercase character followed by any number of alphanumerci characters and underscores (as regular expression `[a-zA-Z][a-zA-Z0-9_]*`), which does not allowed dots and it can't start with number.

So, it must be some proprietary, nonstandard extension.

Thomas2908, 2020-03-05:
You're right. It is probably nonstandard. I didn't know that.
Thank you for you detailed explanation.
Have a nice day!
Tomas

closed 2020-03-05 by j123b567