j123b567/scpi-parser #116: __STDC_VERSION__ macro not always defined

ilanbiala, 2020-09-02:
I'm trying to use this library in a project with g++, and it seems that `__STDC_VERSION__` is not defined when compiling C++ (https://gcc.gnu.org/onlinedocs/cpp/Standard-Predefined-Macros.html). This seems like it poses an issue where:
1. https://github.com/j123b567/scpi-parser/blob/master/libscpi/inc/scpi/cc.h#L44-L55 does not define `C99`
2. https://github.com/j123b567/scpi-parser/blob/master/libscpi/inc/scpi/cc.h#L107-L109 then does not define `HAVE_STDBOOL`
3. https://github.com/j123b567/scpi-parser/blob/master/libscpi/inc/scpi/cc.h#L161-L163 then defines `HAVE_STDBOOL` to be 0
4. https://github.com/j123b567/scpi-parser/blob/ef7112d28859ff50f5b1d870fe933579e1d1fd49/libscpi/inc/scpi/types.h#L53-L55 then defines a typedef for `bool`, which causes a redeclaration error in C++:
```
../libscpi/inc/scpi/types.h:54:27: error: redeclaration of C++ built-in type 'bool' [-fpermissive]
   54 |     typedef unsigned char bool;
      |
```

Should `__cplusplus` be used instead if it is defined to determine which language standard to define for the rest of `cc.h`? `__cplusplus` should also have values like `199711L` and `201103L`.

j123b567, 2020-09-02:
You can always define `HAVE_STDBOOL=1` yourself - on command line or in `scpi_user_config.h` so you can try if it fixes the compilation.

Feel free to provide PR which solve this.

MisterHW, 2021-07-28:
I think I'm getting away with using
```
#if (!HAVE_STDBOOL) && (!_GLIBCXX_HAVE_STDBOOL_H)
    typedef unsigned char bool;
#endif
```

j123b567, 2023-01-16:
C++98 should have stdbool.h so should be fixed by #139

j123b567, 2023-01-16:
Please reopen if it is still an issue.