j123b567/scpi-parser #122: Possible to have multiple, independent scpi session?

mammutserver, 2021-03-16:
Is it possible to have multiple clients connected to a STM32 where each client has his own scpi session?
In my case all clients get an error message if one client sends a wrong command

j123b567, 2021-03-16:
It is not supported in the library itself, but you can have multiple `scpi_t` contexts. Each context has different input buffer, but they can share status nad commands.

Real implementation depends on the usage, connection types, number of sessions, dynamic memory allocation, etc.

mammutserver, 2021-03-17:
So I assume I have to use a scpi_t context for each session and I have to run SCPI_Init for each context?

j123b567, 2021-03-17:
It should be good start. You can share almost everything except input buffer, which you intentionally need separate for each session. You also can't share `error_queue_data` because of the way, how `SCPI_Init` works. You will probably loos all errors after you disconnect.

New client connection heavily depend on you communication scheme. E.g. if you are using TCP, then when you "accept" the connection, then it is the new session.

I don't know if SCPI was ever meant to be used by multiple masters at the same time. There are lot of global and local states which mix together. So in general, it will produce more problems outside this library - locking resources, ...

mammutserver, 2021-03-18:
Thank you @j123b567 
Multiple SCPI sessions over network / TCP are working just fine. Two connections over USB don't seem to work which is understandable.

closed 2021-11-09 by j123b567