j123b567/scpi-parser #136: USE_FULL ERROR LIST report "Unknow Error" for all XE errors

dlock8, 2022-10-05:
I want to use the Full error list. When i push error, the error number reported is good but the string returned is always: Unknow Error

Example:

     int16_t err;
    err=SCPI_ERROR_SYNTAX;  //  Part of the group XE
     fprintf(stdout, "**ERROR: %d, \"%s\"\n\r", (int16_t) err, SCPI_ErrorTranslate(err));
    err=SCPI_ERROR_EXECUTION_ERROR;
     fprintf(stdout, "**ERROR: %d, \"%s\"\n\r", (int16_t) err, SCPI_ErrorTranslate(err));

The printout received is:
 **ERROR: -102, "Unknown error"
 **ERROR: -200, "Execution error"

For error -102, the  expected message to receive is  "Syntax error" and not "Unknown Error". Only errors on group X work fine, any errors on group XE will report the same string: "Unknown Error"

I have try with my own list of user errors with the same results.

What i am doing wrong?

ZTT300, 2023-03-15:
you need "#define USE_FULL_ERROR_LIST  1", because 
` const char * SCPI_ErrorTranslate(int16_t err) {
    switch (err) {
#define X(def, val, str) case def: return str;
**#if USE_FULL_ERROR_LIST
#define XE X**
#else
#define XE(def, val, str)
#endif
        LIST_OF_ERRORS

#if USE_USER_ERROR_LIST
        LIST_OF_USER_ERRORS
#endif
#undef X
#undef XE
        default: return "Unknown error";
    }
}

dlock8, 2023-03-15:
thanks for the info but adding USE_FULL_ERROR_LIST 1  is not enough for get the list of error.  

I have tried all method explained on this case without success:
https://github.com/j123b567/scpi-parser/issues/107

I need to pass the information USE_FULL_ERROR_LIST 1 to the compiler to have the X-macro to be processed.  

For now, my makefile perfom the action below without processing the X-macro.

My final goal is to include my own list of errors USE_USER_ERROR_LIST 1.

Makefile.txt
{{{{
add_definitions(-DSCPI_USER_CONFIG=1)  #DL  flag to add scpi_user_config.h

# Build the SCPI parsing library (a Git submodule) as an external project.
# We'll build it in its source directory using its own Makefile, then place the
# .a file and the library's headers in scpi-parser for including/linking it.
include(ExternalProject)
set(scpi_parser_SOURCE_DIR "${CMAKE_CURRENT_SOURCE_DIR}/lib/scpi-parser/libscpi")
set(scpi_parser_PREFIX "${CMAKE_CURRENT_BINARY_DIR}/scpi-parser")
ExternalProject_Add(scpi_parser_build
	SOURCE_DIR "${scpi_parser_SOURCE_DIR}"
	BUILD_IN_SOURCE ON
	CONFIGURE_COMMAND make clean
	# Only generate the static version of the library, not the shared one (doesn't make sense here).
	BUILD_COMMAND env  "CC=${CMAKE_C_COMPILER}" "CFLAGS=${CMAKE_C_FLAGS}" "PREFIX=${scpi_parser_PREFIX}" make static
	INSTALL_DIR "${scpi_parser_PREFIX}"
	INSTALL_COMMAND cp -r "${scpi_parser_SOURCE_DIR}/dist/libscpi.a" "${scpi_parser_SOURCE_DIR}/inc" "${scpi_parser_PREFIX}"
)
}}}}}

Hope that's help to understand my problem.