j123b567/scpi-parser #138: Termination for SCPI commands

abhimanyukk-druck-utthunga, 2022-11-23:
When I executed VXI11 example, and connected over some VXI11 client codes, we get to know that the commands only work with the termination `\n\r`. So, is there any simple way to avoid this, and work without this termination characters?
Without termination, on the server code, it is showing the below error.
```bash
double free or corruption (out)
Aborted
```

dlock8, 2022-12-05:
I suggest to check the parameter:  SCPI_LINE_ENDING  and file scpi_user_config.h.

for compile the file scpi_user_config.h, you need to add:

add_definitions(-DSCPI_USER_CONFIG=1) 

on CMakelists.txt

Warning: I'm just a user of the scpi parser software and not a devellopper

MisterHW, 2023-01-06:
+1 for `scpi_user_config.h`, which is included in `config.h` to supersede:

https://github.com/j123b567/scpi-parser/blob/224f7cee7c2c5592081e1dec640862f029323782/libscpi/inc/scpi/config.h#L57

hattesen, 2023-09-02:
You might consider using a flexible/dynamic definition of `SCPI_LINE_ENDING` similar to how Keysight is handling it in their [Keysight B2980B Series Femto/Picoammeter Electrometer/High Resistance Meter](https://www.keysight.com/us/en/assets/9921-01475/programming-guides/B2980B-Femto-Picoammeter-and-Electrometer-High-Resistance-Meter-SCPI-Command-Reference.pdf):

**When [parsing commands](https://www.keysight.com/us/en/assets/9921-01475/programming-guides/B2980B-Femto-Picoammeter-and-Electrometer-High-Resistance-Meter-SCPI-Command-Reference.pdf):**

> *Message Terminator*
> A terminator informs SCPI that it has reached the end of a message. The following
> message terminators are permitted.
> * newline <NL>, expressed in ASCII as decimal 10 or hex 0A
> * end or identify <END> (EOI with ATN false)
> * both of the above <NL><END>
> * also <CR><NL>

... although I would recommend ignoring `<END>`, leaving `<NL>` and `<CR><NL>`, only.

**When [returning results](https://www.keysight.com/us/en/assets/9921-01475/programming-guides/B2980B-Femto-Picoammeter-and-Electrometer-High-Resistance-Meter-SCPI-Command-Reference.pdf):**

>*Data Output Format*
> A terminator <newline> (0x0a, 1 byte) is attached to the end of each data.

closed 2023-09-04 by abhimanyukk-druck-utthunga