j123b567/scpi-parser #146: vxi11.h missing ??

roman65536, 2023-02-16:
Hi,

I tried to compile the vxi11 example and looks like either the vxi11.h is missing or the rpc source file, which the vxi11.h will be generated from... Or I missed something :)

rgds
roman

shiwen0536, 2025-11-18:
Hello,
I'm just doing vxi11, randomly cut a vxi11.h file, which can be compiled through.

```
/*
 * Please do not edit this file.
 * It was generated using rpcgen.
 */

#ifndef _VXI11_H_RPCGEN
#define _VXI11_H_RPCGEN

#include <rpc/rpc.h>

#ifdef __cplusplus
extern "C" {
#endif
#define GENERATION_USE_MAIN 0 //main主程序支持
#define GENERATION_USE_SO 1 //so文件支持

#define vxi_printf printf //if(0) printf
#define vxi_fprintf fprintf //if(0) fprintf
//Error states
#define ERR_NO_ERROR  0
#define ERR_SYNTAX_ERROR  1
#define ERR_DEVICE_NOT_ACCESSIBLE  3
#define ERR_INVALID_LINK_IDENTIFIER  4
#define ERR_PARAMETER_ERROR  5
#define ERR_CHANNEL_NOT_ESTABLISHED  6
#define ERR_OPERATION_NOT_SUPPORTED  8
#define ERR_OUT_OF_RESOURCES  9
#define ERR_DEVICE_LOCKED_BY_ANOTHER_LINK  11
#define ERR_NO_LOCK_HELD_BY_THIS_LINK  12
#define ERR_IO_TIMEOUT  15
#define ERR_IO_ERROR  17
#define ERR_INVALID_ADDRESS  21
#define ERR_ABORT  23
#define ERR_CHANNEL_ALREADY_ESTABLISHED  29

#define ERR_PROGRAM_BUG  40

typedef long Device_Link;

enum Device_AddrFamily {
	DEVICE_TCP = 0,
	DEVICE_UDP = 1,
};
typedef enum Device_AddrFamily Device_AddrFamily;

typedef long Device_Flags;

typedef long Device_ErrorCode;

struct Device_Error {
	Device_ErrorCode error;
};
typedef struct Device_Error Device_Error;

struct Create_LinkParms {
	long clientId;
	bool_t lockDevice;
	u_long lock_timeout;
	char *device;
};
typedef struct Create_LinkParms Create_LinkParms;

struct Create_LinkResp {
	Device_ErrorCode error;
	Device_Link lid;
	u_short abortPort;
[... 312 lines cut ...]
extern bool_t xdr_Device_ErrorCode ();
extern bool_t xdr_Device_Error ();
extern bool_t xdr_Create_LinkParms ();
extern bool_t xdr_Create_LinkResp ();
extern bool_t xdr_Device_WriteParms ();
extern bool_t xdr_Device_WriteResp ();
extern bool_t xdr_Device_ReadParms ();
extern bool_t xdr_Device_ReadResp ();
extern bool_t xdr_Device_ReadStbResp ();
extern bool_t xdr_Device_GenericParms ();
extern bool_t xdr_Device_RemoteFunc ();
extern bool_t xdr_Device_EnableSrqParms ();
extern bool_t xdr_Device_LockParms ();
extern bool_t xdr_Device_DocmdParms ();
extern bool_t xdr_Device_DocmdResp ();
extern bool_t xdr_Device_SrqParms ();

#endif /* K&R C */

#ifdef __cplusplus
}
#endif

#endif /* !_VXI11_H_RPCGEN */
```

j123b567, 2025-11-18:
@shiwen0536 Do you have also the rpc source file which can be used to generate this with `rpcgen` and steps to generate it?

j123b567, 2025-11-18:
Ok, so you are using "this" vxi11.x. The problem here is the license. I don't want to incorporate it in the project, but I found the original VXI-11 specification. There are similar code fragments and they stated there that "This document is in the public domain. Permission is granted to reproduce and distribute this document by any means for any purpose." So it is ok for me to copy those fragments from there and compose my own vxi11.x.