j123b567/scpi-parser #158: Parsing decimal values

hakimoune, 2024-04-21:
Hello,
I am having a problem sending decimal value like 2.8 to an STM32 DAC, it worked for me only with integer values. for a DAC in 12 bits mode for example,  accepted values are only from 0 to 4096.
Thank you

j123b567, 2024-05-07:
Hard to say what are you trying to do without example code

hakimoune, 2024-05-07:
Hello Jan,
for example for this code 

static scpi_result_t DAC_MeasureVoltage(scpi_t * context) {
	uint32_t param1;  // Channel number
	double output1 ;

	if (!SCPI_ParamUInt32(context, &param1, TRUE)) {
				return SCPI_RES_ERR;
			}

	if (param1 <= 0 ) {
		return SCPI_ERROR_ILLEGAL_PARAMETER_VALUE;
	} else if (param1 > 2) {
		return SCPI_ERROR_ILLEGAL_PARAMETER_VALUE;
	} else {

		if (param1 == 1) {   // Channel number
			//Channel 1 : PA4
			HAL_DAC_Start(&hdac, DAC_CHANNEL_1);
			output1 = HAL_DAC_GetValue(&hdac, DAC_CHANNEL_1);
		} else {
		//Do something else
		}
	}

	//SCPI_ResultDouble(context, 3.3*output/4095);
	SCPI_ResultDouble(context, output1);
	   return SCPI_RES_OK;
}

it work for integer values from 0 to 4096 but the instruction 
SCPI_ResultDouble(context, 3.3*output/4095);
want result  in decimal values like 2.3

Thank you

jancumps, 2024-05-10:
@hakimoune , have you tried to put 3.3*output/4095 in a variable of type double, and see what its value is?

suggestion to see what's wrong:

```
double dacval = 3.3*output/4095; // set breakpoint for value inspection with debugger
SCPI_ResultDouble(context, dacval);
return SCPI_RES_OK;
```

j123b567, 2024-05-10:
I can't se anything wrong except the typo `output` vs `output1` as pointed out by **jancumps**

hakimoune, 2024-05-12:
It didn't work.

hinxx, 2024-08-15:
@hakimoune the `DAC_ConfigureVoltage` you posted has several issues. Here is a version of it that will set the `var` variable to a correct `uint32_t` value (this is what `HAL_DAC_SetValue()` expects).

```
static scpi_result_t DAC_ConfigureVoltage(scpi_t * context) {
    uint32_t param1; // Channel number
    double param2; // Value 3.3 max

    /* read first parameter if present */
    if (!SCPI_ParamUInt32(context, &param1, TRUE)) {
        return SCPI_RES_ERR;
    }

    /* read second parameter if present */
    if (!SCPI_ParamDouble(context, &param2, TRUE)) {
        return SCPI_RES_ERR;
    }

    uint32_t var = param2*4095/3.3;

    if (param1 <= 0 ) {
          return SCPI_RES_ERR;
        } else if (param1 > 2) {
          return SCPI_RES_ERR;
        } else {

        	if (param1 == 1) {   // Channel number

        		
        		HAL_DAC_SetValue(&hdac, DAC_CHANNEL_1, DAC_ALIGN_12B_R, var);
        		HAL_DAC_Start(&hdac, DAC_CHANNEL_1);
        		
			} else {

				//.........
			}

        }

    return SCPI_RES_OK;
}

```